prampec/IotWebConf #5: Switch off and on the wifi

gergokoos, 2018-09-09:
Hello,

probably it's a weird thing to do, but in my project I don't want to use the wifi all time long, just for some time to set the time up and change the settings.

Is there any way to switch off the wifi (without connecting to the local network or creating an ap) and then switch it on again?

Thanks!

prampec, 2018-11-08:
IotWebConf should be flexible for developers, an option like this can be certainly implemented.

1TekPro, 2019-02-07:
This should be done at sketch level not with IotWebConf.

prampec, 2019-07-01:
A new skipApStartup mode was introduced. So WiFi will be stayed on as long as it is available. However you still cannot force the module from code to jump into AP mode.

1TekPro, 2019-07-01:
Sure you can. Why don't you think it can be done?

prampec, 2019-07-02:
No, no. I meant the IotWebConf does not yet provide a method to jump into AP mode (IOTWEBCONF_STATE_AP_MODE).

hizzawaseem, 2019-10-08:
Hi,
I am relatively new at firmware development. In my project I want the device to start up in Station mode (instead of AP Mode) if device startup configurations are present in EEPROM. At the current instance the device always starts up in AP mode.
I tried to forcefully switch the device to station mode using these commands but it didn't work. 
  WiFi.softAPdisconnect(true);
  WiFi.mode(WIFI_STA);
Thanks!

pfeno, 2019-10-09:
Do you think it's difficult to implement in a futur release ?

prampec, 2019-10-10:
This issue is still open, and I suppose to provide a solution for that. The only thins is that is does not have high priority. :(

dopyoman, 2019-11-25:
Ya I  was looking for a way to do this with the library. It makes me sad that it's not possible.

prampec, 2021-04-14:
This issue is still on the TODO list. Also requested by others #149.

prampec, 2021-06-12:
I have provided to you an OffLine mode. Please check example IotWebConf16OffLineMode.
Solution will be available from v3.1.0

closed 2021-06-12 by prampec