prampec/IotWebConf #14: Start With State 3 if wifi is configured.

AakashKumar21, 2018-12-29:
It takes a lot of time (30 sec approx ) to connect to wifi (state 3) after booting. It should start with state 3.

AakashKumar21, 2018-12-29:
How to i change time for ap mode at start.

prampec, 2018-12-29:
Please consult the IotWebConf.h. I recommend to use the setApTimeoutMs() at startup, and later reset it to a valid value. However it would be a nice feature, to skip AP mode at startup and enter only when connection fails. I note that.

gusarov, 2019-01-07:
I suppose it will work like this if i set ap timout to zero, right?

prampec, 2019-01-07:
Yes. This is my expectation as well.

gusarov, 2019-01-08:
Looks like it is not the case
Just faced a situation when it can't connect...
[image]
And it is not falling back to AP. Well, it is, but looks like 0 timeout forces mode 3 again immediately

prampec, 2019-01-14:
Right.
1. You need to call the setApTimeoutMs in setup() after init was done. (That you did right.)
2. In the loop, you need to reset the timeout by calling the setApTimeoutMs a second time. But please note, that you need to wait some a second to do this, so that IotWebConf enters state 3.

closed 2019-01-14 by prampec