prampec/IotWebConf #21: add a firmware update progress bar

1TekPro, 2019-02-05:
I wrote some code to add a progress bar when using an OLED display on an esp32. How do I go about adding it so it displays when a user clicks "upload"?
I'm guessing somewhere between:
`if(upload.status == UPLOAD_FILE_START){`
`if(Update.end(true)){ `

prampec, 2019-02-05:
While HTTPUpdateServer shouldn't be part of the IotWebConf code, currently it is on ESP32 (because it does hot have an official implementation for ESP32).
But, yes. If you inspect the code, you can see, that in debug mode the software writes some dots to the serial console, when an upload chunk is received (UPLOAD_FILE_WRITE).
At that point you might able operate with HTTPUpload.currentSize and HTTPUpload.totalSize to display your progress bar. (I'm not sure tho, HTTPUpdateServer is not my code.)

closed 2019-02-05 by prampec

1TekPro, 2019-02-07:
@prampec  I am really struggling with this.
I have tried both HTTPUpload,currentSize and upload.currentSize, both error.
If it prints dot, it should print some math...
Serial.printf("Progress: %u%%\r", (upload.currentSize / (upload.totalSize / 100)));

Please, any help would be greatly apprciated.