prampec/IotWebConf #23: Error in AP mode

b1gmans, 2019-02-14:
Everything was working fine, but now I have this error, when I load up a new module or if I force an existing one back into Ap mode by connecting to it from my phone. 

E (27188) event: mismatch or invalid event, id=63
E (27189) event: default event handler failed!

It seems to be something to do with the ESP32 libraries themselves, but I just cannot get any of my projects that I have build with the IotWebConf to work anymore if the device is in AP mode?

Anyone got a work around?

gyromike, 2019-02-15:
I had to let it erase all content when doing the flash update

gyromike, 2019-02-15:
only works the first time.  after that it blocks and waits

gyromike, 2019-02-15:
looks like there was a arduino core change that is causing file problems

prampec, 2019-02-17:
Can you please provide some reference some if you found any connection with other issues?!

gyromike, 2019-02-17:
Still not sure what causes the problem.  I am getting data corruption in my files.  I write a value and then read it back to find that it has changed without me writing to the file again.  It does not look like I can cause the problem with a smaller sketch.  Not sure what to try next.

closed 2019-07-09 by prampec