prampec/IotWebConf #24: Doesn't work with async web server

jeppo77, 2019-02-17:
Are there any plans to make this work with an async web server?

prampec, 2019-02-17:
Please explain what you mean on "async web server"!

jeppo77, 2019-02-17:
sorry maybe I should have been more clear. Please have a look here 
https://github.com/me-no-dev/ESPAsyncWebServer
There is no "handleClient" in the loop. Server connections are handled asynchronously (faster, more connections...)

prampec, 2019-02-17:
Thank you for the idea. This is certainly something that worth considering. At a first look my concern is, that this way we would introduce two third party library dependency. On the other hand, I see the benefits. A solution could be to introduce a fork with async web server.
I keep this thread open.

gibo77, 2019-04-20:
I will try this one.
https://github.com/alanswx/ESPAsyncWiFiManager

prampec, 2019-07-01:
Unfortunatelly ESPAsyncWebServer and WebServer does not share any common features. So it is not obvoius how handle both of them in the same code. So we stuck with the option of forking IotWebConf with using the ESPAsyncWebServer.

prampec, 2021-04-14:
As mentioned in #133, IotWebConf v3.0.0 introduces some nice features to implement this option, however I lack the resources to do it myself. Help is welcomed!

closed 2021-04-14 by prampec