prampec/IotWebConf #25: feature request

aandroide, 2019-02-21:
hello, I would like to use your iot, to control an LED via mqtt and adafruit.
Adafruit requires 4 parameters to add to the Ap configuration:
server name, port. user and aiokey.
in your examples I did not find the possibility to change the mqtt port.
what is the correct method to add these parameters?
Thank you for your time

aandroide, 2019-02-21:
`#include <IotWebConf.h>
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"

// -- Initial name of the Thing. Used e.g. as SSID of the own Access Point.
const char thingName[] = "IOT-Esp-8266";

// -- Initial password to connect to the Thing, when it creates an own Access Point.
const char wifiInitialApPassword[] = "iotesp8266";

#define STRING_LEN 128
#define LED D6  

// -- Configuration specific key. The value should be modified if config structure was changed.
#define CONFIG_VERSION "Adafruit"

// -- When CONFIG_PIN is pulled to ground on startup, the Thing will use the initial
//      password to buld an AP. (E.g. in case of lost password)
#define CONFIG_PIN D2

// -- Status indicator pin.
//      First it will light up (kept LOW), on Wifi connection it will blink,
//      when connected to the Wifi it will turn off (kept HIGH).
#define STATUS_PIN LED_BUILTIN

// -- Callback method declarations.
void wifiConnected();
void configSaved();
boolean formValidator();
void messageReceived(String &topic, String &payload);

// WiFiFlientSecure for SSL/TLS support
WiFiClientSecure client;

DNSServer dnsServer;
WebServer server(80);
HTTPUpdateServer httpUpdater;

char nomeserverValue[STRING_LEN];
char portaValue[STRING_LEN];
char usernameValue[STRING_LEN];
char aiokeyValue[STRING_LEN];

 
IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
IotWebConfParameter nomeserver = IotWebConfParameter("Server MQTT", "Server", nomeserverValue, STRING_LEN);
IotWebConfParameter porta = IotWebConfParameter("Porta MQTT", "Porta", portaValue, STRING_LEN);
IotWebConfParameter username = IotWebConfParameter("Username MQTT", "username", usernameValue, STRING_LEN);
IotWebConfParameter aiokey = IotWebConfParameter("Aio KEY MQTT", "aiokey", aiokeyValue, STRING_LEN);

// Setup the MQTT client class by passing in the WiFi client and MQTT server and login details.
Adafruit_MQTT_Client mqtt(&client, nomeserverValue, portaValue, usernameValue, aiokeyValue);

Adafruit_MQTT_Subscribe onoffbutton = Adafruit_MQTT_Subscribe(&mqtt, usernameValue "/feeds/onoff");

// io.adafruit.com SHA1 fingerprint
 char*fingerprint = "77 00 54 2D DA E7 D8 03 27 31 23 99 EB 27 DB CB A5 4C 57 18";

void MQTT_connect();
void verifyFingerprint();

void setup() 
{
  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW);
  
  Serial.begin(115200);
  Serial.println();
  Serial.println("Starting up...");

[... 165 lines cut ...]
}
void MQTT_connect() {
  int8_t ret;

  // Stop if already connected.
  if (mqtt.connected()) {
    return;
  }

  Serial.print("Connecting to MQTT... ");

  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
       Serial.println(mqtt.connectErrorString(ret));
       Serial.println("Retrying MQTT connection in 5 seconds...");
       mqtt.disconnect();
       delay(5000);  // wait 5 seconds
       retries--;
       if (retries == 0) {
         // basically die and wait for WDT to reset me
         while (1);
       }
  }
  Serial.println("MQTT Connected!");
}`

aandroide, 2019-02-21:
the erro is: invalid conversion from 'char*' to 'uint16_t {aka short unsigned int}' [-fpermissive]

aandroide, 2019-02-21:
furthermore, as soon as I create a new parameter, the first time I find fifty characters printed

prampec, 2019-02-22:
You shouldn't use uninitialized values for variable construction at global scope (e.g. in Adafruit_MQTT_Client mqtt(...) ). At that point in the code, the values are not ready to be used. It might happen that Adafruit_MQTT_Client is not a capable for dynamic configuration.

Besides that, please consult example [03](https://github.com/prampec/IotWebConf/blob/master/examples/IotWebConf03CustomParameters/IotWebConf03CustomParameters.ino) for using numeric parameters.

closed 2019-02-22 by prampec