prampec/IotWebConf #28: how to get IP Address

knopserl, 2019-03-03:
I like and use this library.
There is one feature I miss or at least I don't know how it works:
Before I used the WiFi class which supports the WiFi.isavailable and IPAddress function.
I need to print out the IP Address once it's either connected or it opens an access point (...).
I have seen that I can register a callback once it's connected but how to get the IPAddress?
Which function do I have to use/call to get the IPAddress?

feder-ico, 2019-03-16:
Hi,
You can use WiFi.localIP()

For instance: 
`String myIP = WiFi.localIP().toString();`
`String mySubNetMask = WiFi.subnetMask().toString();`
`String myGateWay"] = WiFi.gatewayIP().toString();`

I hope this helps.

knopserl, 2019-03-17:
@feder-ico thanks, I knew that with th WiFi.localIP() and used it already, it more the question how to do it independent if the thing is in station or AP mode. The is no callback (at least I have not found it) when the AP mode is called to set the IP there.

I solved it in that way meanwhile in the setup() section:
Maybe there is an easier way to do it

void setup()
{
....
  if (isWifiConnected)
  {
    myIP = WiFi.localIP().toString();  
    Serial.println("myIP: " + myIP);
  }
  else
  {
    iotWebConf.doLoop();
    if (needReset)
    {
      Serial.println("Rebooting after 1 second.");
      iotWebConf.delay(1000);
      ESP.restart();
    }
    myIP = WiFi.softAPIP().toString();
    Serial.println("myIP: " + myIP + "(AP mode)");
  }
...
}

closed 2019-03-17 by knopserl