prampec/IotWebConf #29: Dropdown menu for custom parameters?

jessel92, 2019-03-14:
Hello all!

This is either a feature request or just a request for help.
I am by no means an expert on this stuff, so I'm sorry in advance if this is a stupid question hahaha.

So, for one of the custom string parameters id like to create a drop-down menu with only a few specific options, rather than just a text field. Or really any menu that allows people to choose from only a few options rather than having to type it in.

Is there a simple way to do this? Or is this a feature you would be willing to add?

Any help would be greatly appreciated!

prampec, 2019-03-18:
Right! There is a way to do that (however you are right, that there is no example for that).
If you specify a customHtml for the parameter and do not provide a label, that the lib will render the whole customHtml provided by you as the HTML field instead of rendering the input tag automatically.
In your case you should specify your own HTML field in customHtml using the SELECT HTML tag, and omit the label field of the parameter. Specification of the SELECT tag: https://www.w3schools.com/html/html_form_elements.asp

feder-ico, 2019-03-18:
Great!
So, I gess that the following will do the job?

`IotWebConfParameter sayYesOrNoParam = IotWebConfParameter(NULL, "sayYesOrNo", sayYesOrNoValue, NUMBER_LEN, "number", "0", "0", "Say Yes or No: <select name=\"sayYesOrNo\"><option value=\"0\">No</option><option value=\"1\">Yes</option></select>", true);`

But in this case always the default value (No = 0, the first) will be shown and the current value will not. And if the drop-down menu is not used by the user, the parameter will be always overwritten with the default value instead of persisting the current.
Do you imagine a way to show the current value of the parameter as the "selected" value of the select tag?

prampec, 2019-03-18:
I'm afraid you must always place a "selected" string to the end of the item currently selected manually in the Arduino code. Thus always replace the customHtml string with the actual value. (Not very convenient, I know.)

jessel92, 2019-03-19:
YES!! 
That works! Thank you both!
But yeah, since it does not default to the current parameter, it could cause some significant issues for me. 
Do either of you know a way to work around this? 
Any examples would be greatly appreciated.

1TekPro, 2019-05-24:
Set the "selected attribute" as a variable that is checked on page load within the html.

mitchsf, 2019-06-13:
I would really appreciate it if someone would post a complete example. I don't know enough HTML to figure it out from the conversation. Thanks!

mitchsf, 2019-06-29:
Hopefully this will happen. Several of my projects have over 40 parameters to configure. Only two or three are text fields, the rest are choices, so a dropdown works best. Even better would be the ability to select a variety of objects, such as volume controls, radio buttons, etc.

prampec, 2019-07-09:
Please note, that HTML customization of the config portal is now available from v2.3.0.

mitchsf, 2019-07-10:
Thanks for your reply. The issue is, that there are no examples, and I don't know enough HTML to figure it out from this conversation. Below is as far as I got. The dropdown is diplayed, but I don't know how to set a default, and how to capture the response.

`/**
 * IotWebConf03CustomParameters.ino -- IotWebConf is an ESP8266/ESP32
 *   non blocking WiFi/AP web configuration library for Arduino.
 *   https://github.com/prampec/IotWebConf 
 *
 * Copyright (C) 2018 Balazs Kelemen <prampec+arduino@gmail.com>
 *
 * This software may be modified and distributed under the terms
 * of the MIT license.  See the LICENSE file for details.
 */

/**
 * Example: Custom parameters
 * Description:
 *   In this example it is shown how to attach your custom parameters
 *   to the config portal. Your parameters will be maintained by 
 *   IotWebConf. This means, they will be loaded from/saved to EEPROM,
 *   and will appear in the config portal.
 *   Note the configSaved and formValidator callbacks!
 *   (See previous examples for more details!)
 * 
 * Hardware setup for this example:
 *   - An LED is attached to LED_BUILTIN pin with setup On=LOW.
 *   - [Optional] A push button is attached to pin D2, the other leg of the
 *     button should be attached to GND.
 */

#include <IotWebConf.h>

// -- Initial name of the Thing. Used e.g. as SSID of the own Access Point.
const char thingName[] = "Thing";

// -- Initial password to connect to the Thing, when it creates an own Access Point.
const char wifiInitialApPassword[] = "TigerDog";

#define STRING_LEN 128
#define NUMBER_LEN 32

// -- Configuration specific key. The value should be modified if config structure was changed.
#define CONFIG_VERSION "dem2"

// -- When CONFIG_PIN is pulled to ground on startup, the Thing will use the initial
//      password to buld an AP. (E.g. in case of lost password)
#define CONFIG_PIN 34

// -- Status indicator pin.
//      First it will light up (kept LOW), on Wifi connection it will blink,
//      when connected to the Wifi it will turn off (kept HIGH).
#define STATUS_PIN 17

// -- Callback method declarations.
void configSaved();
boolean formValidator();

DNSServer dnsServer;
WebServer server(80);

char stringParamValue[STRING_LEN];
char intParamValue[NUMBER_LEN];
char floatParamValue[NUMBER_LEN];
char sayYesOrNoParamValue[NUMBER_LEN];

IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
IotWebConfParameter stringParam = IotWebConfParameter("String param", "stringParam", stringParamValue, STRING_LEN);
IotWebConfSeparator separator1 = IotWebConfSeparator();
IotWebConfParameter intParam = IotWebConfParameter("Int param", "intParam", intParamValue, NUMBER_LEN, "number", "1..100", NULL, "min='1' max='100' step='1'");
IotWebConfParameter floatParam = IotWebConfParameter("Float param", "floatParam", floatParamValue, NUMBER_LEN, "number", "e.g. 23.4", NULL, "step='0.1'");
IotWebConfParameter sayYesOrNoParam = IotWebConfParameter(NULL, "sayYesOrNoParam", sayYesOrNoParamValue, NUMBER_LEN, "number", "0", "0", "Say Yes or No: <select name=\"sayYesOrNo\"><option value=\"0\">No</option><option value=\"1\">Yes</option></select>", true);
void setup() 
{
  Serial.begin(115200);
  Serial.println();
  Serial.println("Starting up...");

  iotWebConf.setStatusPin(STATUS_PIN);
  iotWebConf.setConfigPin(CONFIG_PIN);
  iotWebConf.addParameter(&stringParam);
  iotWebConf.addParameter(&separator1);
  iotWebConf.addParameter(&intParam);
  iotWebConf.addParameter(&floatParam);
  iotWebConf.addParameter(&sayYesOrNoParam);
  iotWebConf.setConfigSavedCallback(&configSaved);
  iotWebConf.setFormValidator(&formValidator);
  iotWebConf.getApTimeoutParameter()->visible = true;

  // -- Initializing the configuration.
  iotWebConf.init();

  // -- Set up required URL handlers on the web server.
  server.on("/", handleRoot);
  server.on("/config", []{ iotWebConf.handleConfig(); });
  server.onNotFound([](){ iotWebConf.handleNotFound(); });

  Serial.println("Ready.");
}

void loop() 
{
  // -- doLoop should be called as frequently as possible.
  iotWebConf.doLoop();
}

/**
 * Handle web requests to "/" path.
 */
void handleRoot()
{
  // -- Let IotWebConf test and handle captive portal requests.
  if (iotWebConf.handleCaptivePortal())
  {
    // -- Captive portal request were already served.
    return;
  }
  String s = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/>";
  s += "<title>IotWebConf 03 Custom Parameters</title></head><body>Hello world!";
  s += "<ul>";
  s += "<li>String param value: ";
  s += stringParamValue;
  s += "<li>Int param value: ";
  s += atoi(intParamValue);
  s += "<li>Float param value: ";
  s += atof(floatParamValue);
  s += "</ul>";
  s += "<li>Dropdown: ";
  s += sayYesOrNoParamValue;
  s += "Go to <a href='config'>configure page</a> to change values.";
  s += "</body></html>\n";

  server.send(200, "text/html", s);
}

void configSaved()
{
  Serial.println("Configuration was updated.");
}

boolean formValidator()
{
  Serial.println("Validating form.");
  boolean valid = true;

  int l = server.arg(stringParam.getId()).length();
  if (l < 3)
  {
    stringParam.errorMessage = "Please provide at least 3 characters for this test!";
    valid = false;
  }

  return valid;
}`

prampec, 2019-07-11:
You are right! This is why this thread is still not closed.

feder-ico, 2019-09-21:
Hi,
Since v2.3.0 includes HTML customization using Javascript, now is possible to drop-down menus for custom parameters.
I took the example @prampec provided for adding the button to password fields and added some code to turn the inputs into selects and use drop-downs menus. At least for numeric parameters.

Let me share...
In the example _examples/IotWebConf10CustomHtml/IotWebConf10CustomHtml.ino_ I have replaced the the following code:
```
\n\
document.addEventListener('DOMContentLoaded', function(event) {\n\
  let elements = document.querySelectorAll('input[type=\"password\"]');\n\
  for (let p of elements) {\n\
  	let btn = document.createElement('INPUT'); btn.type = 'button'; btn.value = '🔓'; btn.style.width = 'auto'; p.style.width = '83%'; p.parentNode.insertBefore(btn,p.nextSibling);\n\
    btn.onclick = function() { if (p.type === 'password') { p.type = 'text'; btn.value = '🔒'; } else { p.type = 'password'; btn.value = '🔓'; } }\n\
  };\n\
});\n
```
With this one that includes the first:
```
document.addEventListener(
        'DOMContentLoaded',
        function (e) {
            let elements = document.querySelectorAll('input[type=\"password\"]');

            for (let p of elements) {
                let btn = document.createElement('INPUT'); btn.type = 'button'; btn.value = '🔓'; btn.style.width = 'auto'; p.style.width = '83%'; p.parentNode.insertBefore(btn,p.nextSibling);

                btn.onclick = function() { if (p.type === 'password') { p.type = 'text'; btn.value = '🔒'; } else { p.type = 'password'; btn.value = '🔓'; } }
            };

            var customSelects = document.querySelectorAll('input[type="custom_select"]');

            if (!customSelects.length) return;

            for (var item = 0; item < customSelects.length; ++ item) {
                var defaultValue = customSelects[item].value;
                    optionsList = customSelects[item].getAttribute('data-options').split('|'),
                    selectTag = [
                        '<select name="',
                        customSelects[item].name,
                        '" id="',
                        customSelects[item].id,
                        '">'
                    ];
					
				if (defaultValue === '') {
					selectTag.push('<option value=""></option>');
				}

                for (var itemOption in optionsList) {
                    selectTag.push('<option value="');
                    selectTag.push(itemOption);
                    selectTag.push('"');

                    if (itemOption == defaultValue) {
                        selectTag.push(' selected');
					}

                    selectTag.push('>');
                    selectTag.push(optionsList[itemOption]);
                    selectTag.push('</option>');
                }

                selectTag.push('</select>');

                var parent = customSelects[item].parentNode,
                    span = document.createElement('span');

                span.innerHTML = selectTag.join('');

                parent.removeChild(customSelects[item]);
                parent.appendChild(span);
            }
        }
    )
```
The minimized line for better include in the sketch is the following:

`document.addEventListener(\"DOMContentLoaded\",function(e){let t=document.querySelectorAll('input[type=\"password\"]');for(let e of t){let t=document.createElement(\"INPUT\");t.type=\"button\",t.value=\"🔓\",t.style.width=\"auto\",e.style.width=\"83%\",e.parentNode.insertBefore(t,e.nextSibling),t.onclick=function(){\"password\"===e.type?(e.type=\"text\",t.value=\"🔒\"):(e.type=\"password\",t.value=\"🔓\")}}var s=document.querySelectorAll('input[type=\"custom_select\"]');if(s.length)for(var o=0;o<s.length;++o){var n=s[o].value;for(var l in optionsList=s[o].getAttribute(\"data-options\").split(\"|\"),selectTag=['<select name=\"',s[o].name,'\" id=\"',s[o].id,'\">'],\"\"===n&&selectTag.push('<option value=\"\"></option>'),optionsList)selectTag.push('<option value=\"'),selectTag.push(l),selectTag.push('\"'),l==n&&selectTag.push(\" selected\"),selectTag.push(\">\"),selectTag.push(optionsList[l]),selectTag.push(\"</option>\");selectTag.push(\"</select>\");var a=s[o].parentNode,p=document.createElement(\"span\");p.innerHTML=selectTag.join(\"\"),a.removeChild(s[o]),a.appendChild(p)}});`

Then, whe defining the parameter, instead of doing this:

`IotWebConfParameter enableMonitoringParam = IotWebConfParameter("Enable Monitoring", "enableMonitoring", enableMonitoringValue, NUMBER_LEN, "number", "0", "0", "min='0' max='3' step='1'", true);`

You should do this:

`IotWebConfParameter enableMonitoringParam = IotWebConfParameter("Enable Monitoring", "enableMonitoring", enableMonitoringValue, NUMBER_LEN, "custom_select", "0", "0", "data-options='No|Yes|I do not know|Wharever'", true);`

Then when displaying the parameters a drop-down menu will appear showing the options defined as `data-options` separated with `|` (pipe).
If the field already has a value it will be showed as selected. If field is empty, the empty option will be added on the drop-down allowing to save the default value for the first time that is saved when field is empty. Next time, that option will be the selected one.
When the user selects an option from the drop-down menu, a numeric value will be saved on the field accordingly. The first option will be saved as 0, the second as 1, the thirds as 2 as so on and so forth.
In the pasted code, "No" will be saved as 0, "Yes" will be saved as 1, "I do not know" will be saved as 2 and "whatever" will be saved as 3.

Hope this helps!

prampec, 2021-04-14:
With v3.0.0 drop-down is introduced for you.
As well as the possibility to customize input controls more nicely.

closed 2021-04-14 by prampec

schmurtzm, 2022-01-06:
Sorry to re-open, I'm a little lost with the right choice to make a drop-down on v3.x...

My objective is to display a drop-down which allow 3 choices and keep it in flash memory (with restore on boot).
An example in this issue is based on IotWebConf10CustomHtml.ino but may "[IotWebConf12CustomParameterType.ino](https://github.com/prampec/IotWebConf/blob/master/examples/IotWebConf12CustomParameterType/IotWebConf12CustomParameterType.ino)" is more adapted to do what I want ?

I still think that a little example of drop-down could be useful 😅