prampec/IotWebConf #31: Default value in config form

feder-ico, 2019-03-17:
Hi,
I have just a question,
It seems that default value is not working on the configuration form when empty.
Can you help me by telling me if I am doing something wrong?
Placeholder text is showing ok but if field is not completed default value is not saved.

`char mqttServerValue[STRING_LEN];`

`IotWebConfParameter mqttServerParam = IotWebConfParameter("MQTT server", "mqttServer", mqttServerValue, STRING_LEN, "text", "mymqttserver.com", "mymqttserver.com", NULL, true);`

`void setup() 
{  iotWebConf.addParameter(&mqttServerParam);
}`

Am I doing something wrong or my expectations are wrong and the behavior is the expected?
What I need to achieve is that **if field is not filled by user, then the default value is saved.**

Tanks in advance for your support.

prampec, 2019-03-17:
Duplicate. See issue #19

closed 2019-03-17 by prampec

feder-ico, 2019-03-18:
Ok, I will try to work on this (with my very very very limited knowledge) 
Let´s see what I can achieve... :)