prampec/IotWebConf #32: Password parameter visibility

pfeno, 2019-03-18:
Is it possible to add the functionality, like in Android (as picture below) and some others OS, to have an icon to make visible the password?

[image]

prampec, 2019-03-18:
Currently IotWebConf config visualization is based on pure HTML without any additional libraries to keep the solution as simple as possible and keep the footage low. As far as I know currently HTML doesn't support the feature you want natively.
The solution could be to add the possibility for including external HTML libraries, but IotWebConf does not support that either. So let's keep this issue open, to fix this one.

prampec, 2019-07-09:
Done in v2.3.0
Please check example 10, that implements exactly this feature.

closed 2019-07-09 by prampec