prampec/IotWebConf #34: Does not play nice with FreeRTOS queues and tasks with Arduino SDK

arrowcircle, 2019-03-26:
Hi! I am trying to run IotWebConf with Arduino IDE using queues and tasks.
But I get errors - serial monitor closes connection.

```
--- Miniterm on /dev/cu.usbserial-1420  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
CAN Receiver
Successfully initialized CAN bus
Config size: 165
Loaded config 'iwcThingName'= 'APNAME'
Loaded config 'iwcApPassword'= '12345678'
Loaded config 'iwcWifiSsid'= 'internet'
Loaded config 'iwcWifiPassword'= '12345678'
Loaded config 'iwcApTimeout'= '30'
State changing from: 0 to 2
0x12: (HEX): 68 65 6C 6C 6F 1A 9B 1E  (DEC): 104 101 108 108 111 26 155 30  (CHAR): h e l l o ␚ � ␞ 
Setting up AP: APNAME
Use password: 12345678
AP IP address: 192.168.4.1
State changed from: 0 to 2
0x12: (HEX): 68 65 6C 6C 6F 1A 9B 1E  (DEC): 104 101 108 108 111
--- exit ---
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/ionbuggy/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 445, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/Users/ionbuggy/.platformio/penv/lib/python2.7/site-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
SerialException: read failed: [Errno 6] Device not configured

Terminal will be reused by tasks, press any key to close it.

```
This message in logs from my code, that outputs received messages to serial: 
```
0x12: (HEX): 68 65 6C 6C 6F 1A 9B 1E  (DEC): 104 101 108 108 111
```

I have few tasks:
* Receives messages from queue and process them (outputs to serial)
* Sends demo data to CAN bus
* Loops iotwebconf every 10 msecs.

Any ideas how to make it all play nice together?

arrowcircle, 2019-03-26:
It looks like it was hardware failure, but thing is it does not play nice with FreeRTOS. Connection to WiFi takes up to 30 secs, and if loop is run with high priority, it just blocks everything. This lib uses delay inside. Is there are any ways to change delay to osDelay of FreeRTOS for ESP32?

prampec, 2019-03-26:
Negative! This library is aimed to avoid using delays, so it does not utilizes delay inside.

arrowcircle, 2019-03-27:
@prampec https://github.com/prampec/IotWebConf/blob/master/src/IotWebConf.cpp#L706

prampec, 2019-03-27:
Right! Using IotWebConf.delay() is highly optional. So in this case, please stop using this option.

closed 2019-03-27 by prampec