prampec/IotWebConf #36: ESP8266HTTPUpdateServer class compile error on Wemos D1 mini

mekanoid, 2019-04-03:
I get an error when compiling IotWebConf in Arduino IDE for Wemos D1 mini. It shows up even when I try to compile your "IotWebConf01Minimal" example from the repository.

I have searched the net without any luck, so now I try this way. It would be nice to solve this, because I really would like to use IotWebConfig instead of WifiManager. ;-)

_And thanks a lot for the good work with IotWebConf !_

```
Arduino:1.8.5 (Linux), Kort:"WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

Build options changed, rebuilding all
/home/peter/Arduino/libraries/IotWebConf/src/IotWebConf.cpp: In member function 'void IotWebConf::stateChanged(byte, byte)':
/home/peter/Arduino/libraries/IotWebConf/src/IotWebConf.cpp:781:30: error: 'class ESP8266HTTPUpdateServer' has no member named 'updateCredentials'
         this->_updateServer->updateCredentials(IOTWEBCONF_ADMIN_USER_NAME, this->_apPassword);
                              ^
Multiple libraries were found for "WiFiUdp.h"
  Used: /home/peter/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi
Unused: /home/peter/Program/arduino-1.8.5/libraries/WiFi
Unused: /home/peter/Program/arduino-1.8.5/libraries/WiFi
Unused: /home/peter/Program/arduino-1.8.5/libraries/WiFi
Unused: /home/peter/Program/arduino-1.8.5/libraries/WiFi
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

prampec, 2019-04-04:
Right. As written in the README file: "For ESP8266, IotWebConf requires the esp8266 board package version 2.4.2 or later!"
So please upgrade your ESP8266 board package!

closed 2019-04-04 by prampec

mekanoid, 2019-04-06:
Hmmm, not very obvious to me that this error had something to do with the board package... :(
But I upgraded to 2.4.2 and now I get most functions compiling. Thank you!

I still get compile errors for example when using:
`IotWebConfSeparator separator1 = IotWebConfSeparator("Test separator");`

prampec, 2019-04-06:
Okay. This is a new feature, and however the this git repo has it in the examples, the "released" versions does not contains this feature. So you have two separate options:
1. Try the examples arriving with the library in the Arduino package manager, and use the "released" features.
2. Do not use the Arduino package manager, but clone this repo to the library folder manually and try the "unreleased" version.

puyiot, 2020-12-22:
I followed the instructions Not sure if it's what you want or not.
`
//esp8266 เวอร์ชัน 2.4.2 หรือใหม่กว่า

#include <IotWebConf.h>

// -- Initial name of the Thing. Used e.g. as SSID of the own Access Point.
const char thingName[] = "PUY IOT FIRST";
const char wifiInitialApPassword[] = "111111111";
#include <TridentTD_LineNotify.h>

String LineText;
String string1 = "ศาลา2 ระบบควบคุมเริ่มทำงาน  MCU RESET ครั้งที่  ";

int blynk_Connect = 0;
int Line1_status = 0;
int Line2_status = 0;

#define Line_Token 45
#define Blynk_Token 34

#define CONFIG_VERSION "dem2"

// -- Callback method declarations.
void configSaved();
boolean formValidator();

#include <DNSServer.h>
DNSServer dnsServer;
#include <ESP8266WebServer.h>
ESP8266WebServer server(80);//Server on port 80
//WebServer server(80);

char Line_Token_1[Line_Token];
//8HBPx7WHUTjGoPWACom1i98WEHGeCr6jZIDFx9L7kry
//QerWyEZdBShylI3iTTxKz1Pk8Zo1vLy2NQa7AXFgYLM
char Blynk_Token_1[Blynk_Token];
//f1XdbTweXh0vGR2Gojy0ltrz99TGsIfh

IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
IotWebConfParameter Token_Line = IotWebConfParameter("Line_token", "Line_token", Line_Token_1, Line_Token);
IotWebConfSeparator separator1 = IotWebConfSeparator();
IotWebConfParameter Token_Blynk = IotWebConfParameter("Blynk_Token", "Blynk_Token", Blynk_Token_1, Blynk_Token);

String SSID_NAME;
String PASSWORD_NAME;
String IP;
String Sub_M;
String Gate_way;

String SSID_AP;
String PASSWORD_AP;

#include <BlynkSimpleEsp8266.h>
//#include <BlynkSimpleEsp32.h>

#define BLYNK_DEBUG
#define BLYNK_PRINT Serial
int blynkIsDownCount = 0;
BlynkTimer timer;

//-------------สำหรับ Server local ที่แจกให้ เพิ่ม **** แค่ 2 บรรทัดนี้--------------
//(ถ้าเป็น Blynk Server ปกติไม่ต้องใส่)  ----------//
char server1[] = "oasiskit.com";
//int port = 8080;

#include <EEPROM.h>

int MCU_Reset = 0;
int count_MCU_Rset;
int Amount_Mcu_Reset;
int reset_MCU;

void setup()
{

  Serial.begin(115200);
  Serial.println();
  Serial.println(" Starting up... ");

  EEPROM.begin(1024);

  iotWebConf.addParameter(&Token_Line);
  iotWebConf.addParameter(&Token_Blynk);

  iotWebConf.setConfigSavedCallback(&configSaved);
  iotWebConf.setFormValidator(&formValidator);
  iotWebConf.getApTimeoutParameter()->visible = true;

  // -- Initializing the configuration.
  iotWebConf.init();

  // -- Set up required URL handlers on the web server.
  server.on("/", handleRoot);
  server.on("/config", [] { iotWebConf.handleConfig(); });
  server.onNotFound([]() {
    iotWebConf.handleNotFound();
    iotWebConf.handleCaptivePortal();
  });

  Serial.println("Ready");
  Serial.println("Finish");

  // Blynk.config(blynk_token);////เริ่มการเชื่อมต่อ Blynk Server แบบปกติ
  Blynk.config(Blynk_Token_1, server1, 8080);
  ////เริ่มการเชื่อมต่อ Blynk Server*********สำหรับ Server local ที่แจกให้ เพิ่ม, server, port  แค่นี้จบ

  timer.setInterval(30000L, reconnecting);  //Function reconnect

  SSID_NAME = iotWebConf.getWifiSsidParameter()->valueBuffer;

  PASSWORD_NAME = iotWebConf.getWifiPasswordParameter()->valueBuffer;

  SSID_AP = iotWebConf.getThingNameParameter()->valueBuffer;

  PASSWORD_AP = iotWebConf.getApPasswordParameter()->valueBuffer;

  LINE.setToken(Line_Token_1);

  count_MCU_Rset = EEPROM.get(150, Amount_Mcu_Reset);
  Serial.print("count_MCU_Rset =  ");
  Serial.println(count_MCU_Rset);

  reset_MCU = count_MCU_Rset + 1;
  //  Serial.print("reset_MCU =  ");
  //  Serial.println(reset_MCU);
  Amount_Mcu_Reset = EEPROM.put(101, reset_MCU);
  EEPROM.commit();
  //  Serial.print("Amount_Mcu_Reset =  ");
  //  Serial.println(Amount_Mcu_Reset);

  count_MCU_Rset = EEPROM.put(150, Amount_Mcu_Reset);
  EEPROM.commit();
  Serial.print("count_MCU_Rset =  ");
  Serial.println(count_MCU_Rset);

}//void setup

void loop()
{

  if (Blynk.connected())
  {
    Blynk.run();

    blynk_Connect = 1;

  } else {

    blynk_Connect = 0;

  }

  timer.run();

  // -- doLoop should be called as frequently as possible.
  iotWebConf.doLoop();

  delay(500);

}

void reconnecting()
{

  if (!Blynk.connected())
  {
    blynkIsDownCount++;
    BLYNK_LOG("blynk server is down! %d  times", blynkIsDownCount);
    Blynk.connect(5000);
  }
}

BLYNK_CONNECTED()
{

}

/**
   Handle web requests to "/" path.
*/
void handleRoot()
{

  Serial.println(" NOW! CONFIG AP ");

  // -- Let IotWebConf test and handle captive portal requests.
  if (iotWebConf.handleCaptivePortal())
  {
    // -- Captive portal request were already served.
    return;
  }
  String s = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/>";
  s += "<title>IotWebConf 03 Custom Parameters</title></head><body>Hello world!";
  s += "<ul>";

  s += "<li>String param value1: ";
  s += Line_Token_1;
  s += "<li>String param value2: ";
  s += Blynk_Token_1;

  s += "</ul>";
  s += "Go to <a href='config'>configure page</a> to change values.";
  s += "</body></html>\n";

  server.send(200, "text/html", s);
}

void configSaved()
{
  Serial.println("Configuration was updated.");

  Serial.println(" SAVE! CONFIG AP");

  Serial.println(" DISCON. CONFIG!");

  SSID_AP = iotWebConf.getThingNameParameter()->valueBuffer;

  PASSWORD_AP = iotWebConf.getApPasswordParameter()->valueBuffer;

}

boolean formValidator()
{
  Serial.println("Validating form.");
  boolean valid = true;

  int l = server.arg(Token_Line.getId()).length();
  if (l < 3)
  {
    Token_Line.errorMessage = "Please provide at least 3 characters for this test!";
    valid = false;
  }

  return valid;
}

`

prampec, 2020-12-22:
Sorry, I don't understand this comment. Is there a problem here? Also note, that GIT version is just upgraded to v3.0.0, please follow migration guide in case you find suspicious errors.