prampec/IotWebConf #37: Device Crashes with interrupt during change of connection state

jessel92, 2019-04-06:
Hi All!
I'm using a high-resolution encoder that uses interrupts to track the length of material traveled. 
I'm running into a significant issue that crashes the device when the encoder is tracking and iotWebconf switches connection state at the same time. 
This happens with both the ESP8266 and the ESP32.
Does anyone have any idea why this would happen?
I can't for the life of me figure it out, and it's super important for my project that this doesn't happen. 
Let me know if there is any info I could give to help! 

Thanks in advance!

prampec, 2019-04-07:
I'm sorry to hear that. IotWebConf built with simplicity in mind. With hoping that simplicity will also mean stability.
Does this happen with each transition, or with just one.
Please provide the serial log!

jessel92, 2019-04-07:
So as soon as I posted this issue, I seemed to have solved the problem with the ESP32!
It appears to be a flash/cache error
If you use the IRAM_ATTR macro in the interrupt handlers, it fixes the issue.

I still have yet to find the solution for the ESP8266.
This error occurs with every state change, but I will get you the error log ASAP.

prampec, 2019-07-01:
Also note that an extra "yield" is added to "doLoop" lately to improve the stability.

closed 2019-07-01 by prampec

ximon, 2019-08-19:
I'm running into this with interrupts caused by a signal every 40ms, whilst the state is changing it will crash with exception(0). Once the state is connected it will run fine for hours. I tried disabling the serial output in iotwebconf.h by commenting out the line #define IOTWEBCONF_DEBUG_TO_SERIAL but this made little difference ( i thought it might have something to do with the interrupt firing whilst writing to serial - but seems not to be)

Serial log: (Starting... is the second line of my setup(), with the first being Serial.begin(115200);
```
Starting...
Config size: 165
Loaded config 'iwcThingName'= 'HotTubRemote'
Loaded config 'iwcApPassword'= <hidden>
Loaded config 'iwcWifiSsid'= 'VM7430922'
Loaded config 'iwcDecoding button command,Temperature down button was pressed
WifiPassword'= <hidden>
Loaded config 'iwcApTimeout'= '30'
HTTP server started.
SkipApStartup mode was applied
State changing from: 0 to 3
Connecting to [VM7430922] (password is hidden)

Exception (0):
epc1=0x402029f4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff9f0 end: 3fffffc0 offset: 01a0
3ffffb90:  4010048e feefeffe feefeffe 40100514  
3ffffba0:  c0037020 feefeffe feefeffe feefeffe  
3ffffbb0:  feefeffe feefeffe feefeffe 00000022  
3ffffbc0:  3fffc200 40100454 3fffc258 4000050c  
3ffffbd0:  400043d3 00000030 00000016 ffffffff  
3ffffbe0:  400044ab 3fffc718 3ffffcd0 08000000  
3ffffbf0:  60000200 08000000 00000003 00000000  
3ffffc00:  0000ffff 00042035 00002035 003fd000  
3ffffc10:  4025a2d0 0000049c 003fd000 00000030  
3ffffc20:  feefeffe feefeffe feefeffe feefeffe  
3ffffc30:  feefeffe feefeffe feefeffe 3fff017c  
3ffffc40:  3ffe8aa2 00000020 00000000 40218a0d  
3ffffc50:  feefeffe 40214ed8 feefeffe feefeffe  
3ffffc60:  feefeffe feefeffe feefeffe feefeffe  
3ffffc70:  feefeffe feefeffe feefeffe 00000000  
3ffffc80:  3ffe8aa2 00000020 3ffffd20 40214ed8  
3ffffc90:  feefeffe feefeffe feefeffe feefeffe  
3ffffca0:  3ffe9351 00000000 3ffe9350 4021806e  
3ffffcb0:  00000000 4000444e 9f7ced91 00000000  
3ffffcc0:  00000002 4010580b 00000001 60000200  
3ffffcd0:  00000002 4000410f 00001001 00000205  
3ffffce0:  3fffc718 40004a3c 000003fd 4025a2d0  
3ffffcf0:  3fffc718 40105a9c 000003fd 4010066c  
3ffffd00:  4024bb34 000003fd 3ffefee8 4024bb1d  
3ffffd10:  3fff1e2c 4024bc07 3fff198c 0000049c  
3ffffd20:  000003fd 3ffffe60 3fff198c 4024bbea  
3ffffd30:  ffffff00 55aa55aa 0000002c 00000020  
3ffffd40:  00000020 00000037 00000037 aa55aa55  
3ffffd50:  000003ff 4024c0d4 3fff198c 3fff198c  
3ffffd60:  00000001 3fff198c 0000049c 00000000  
3ffffd70:  40100271 00000001 3fff199c 4024c2cc  
3ffffd80:  3ffef180 3fff198c 00000001 3ffffe60  
3ffffd90:  3ffffe80 3fff19c3 0000002c 00000020  
3ffffda0:  3fff1a4c 3ffffec1 00000001 4024c39e  
3ffffdb0:  3ffffe60 40259f30 00000000 0000000c  
3ffffdc0:  3fff1d8c 3ffffe80 3fff198c 4024c36d  
3ffffdd0:  3fff198c 4024c3d4 3ffe8524 3ffef6de  
3ffffde0:  4020704e 3ffef6de 3ffef6ff 40206fab  
3ffffdf0:  34374d56 32393033 3fff0032 40218fdc  
3ffffe00:  3fff01ac 3ffefc9c 3fff1844 00000000  
3ffffe10:  6b716438 54597a76 7968636a 40218400  
3ffffe20:  3fff01ac 00000040 00000050 3ffeff18  
3ffffe30:  00000033 0000000a 3ffeff18 40213cac  
3ffffe40:  3ffe88f6 00000000 3ffe88f5 402184de  
3ffffe50:  43d34800 4039ac3e 000001fc 40218744  
3ffffe60:  34374d56 32393033 3fff0032 40218b6c  
[... 42 lines cut ...]
0x40206f73: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 156
0x40218e6c: free(void*) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1764
0x40218300: uart_write_char(uart_t*, char) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\uart.cpp line 476
0x40213b0c: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x4021836e: uart_write(uart_t*, char const*, size_t) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\uart.cpp line 498
0x40218a44: _umm_free(void*) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1304
0x40218e6c: free(void*) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1764
0x40218300: uart_write_char(uart_t*, char) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\uart.cpp line 476
0x40213b0c: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 158
0x4021836e: uart_write(uart_t*, char const*, size_t) at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\uart.cpp line 498
0x40100484: millis() at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 186
0x40204866: IotWebConf::connectWifi(char const*, char const*) at C:\Users\simon\Documents\Arduino\libraries\IotWebConf\src\IotWebConf.cpp line 1112
[... cut ...]
0x402140a8: Print::println() at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\Print.cpp line 178
0x40205e8d: IotWebConf::doLoop() at C:\Users\simon\Documents\Arduino\libraries\IotWebConf\src\IotWebConf.cpp line 775
0x40201e74: loop() at C:\Users\simon\Source\Repos\Hot-tub-remote/Hot-tub-remote.ino line 270
0x40216324: loop_wrapper() at C:\Users\simon\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125
```

Hope this helps!

seugenes, 2019-09-30:
> Hi All!
> I'm using a high-resolution encoder that uses interrupts to track the length of material traveled.
> I'm running into a significant issue that crashes the device when the encoder is tracking [...]

Have you managed to solve this? Thanks in advance.

feder-ico, 2019-09-30:
I faced a similar problem and I workarounded avoiding using interrupts while iotWebConf is not connected.
Since in my project I do nothing local and everithing is about sending information make no sense to use the interruptions while offline so if  `!= IOTWEBCONF_STATE_ONLINE` I deattach interrupt and I attach it again once `== IOTWEBCONF_STATE_ONLINE` 
Hopes this helps.