prampec/IotWebConf #38: Compile errors with new version of Arduino IDE (1.8.9)

pinootto, 2019-04-11:
Hello,
today I updated my Arduino IDE from 1.8.3 to 1.8.9.
On 1.8.3 my soyrce code, which uses IotWebConf. was compiling and working well.
After that upgrade, I get a lot of errors:

Arduino: 1.8.9 (Linux), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Build options changed, rebuilding all
In file included from /home/giovanni/Arduino/libraries/IotWebConf/src/IotWebConf.h:16:0,
                 from /home/giovanni/development/projects/mqtt-ir-control/mqtt-ir-control-01/mqtt-ir-control-01.ino:40:
/home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/DNSServer/src/DNSServer.h:67:13: error: cannot declare field 'DNSServer::_udp' to be of abstract type 'WiFiUDP'
     WiFiUDP _udp;
             ^
In file included from /home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/DNSServer/src/DNSServer.h:3:0,
                 from /home/giovanni/Arduino/libraries/IotWebConf/src/IotWebConf.h:16,
                 from /home/giovanni/development/projects/mqtt-ir-control/mqtt-ir-control-01/mqtt-ir-control-01.ino:40:
/opt/arduino/libraries/WiFi/src/WiFiUdp.h:27:7: note:   because the following virtual functions are pure within 'WiFiUDP':
 class WiFiUDP : public UDP {
       ^
In file included from /opt/arduino/libraries/WiFi/src/WiFiUdp.h:23:0,
                 from /home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/DNSServer/src/DNSServer.h:3,
                 from /home/giovanni/Arduino/libraries/IotWebConf/src/IotWebConf.h:16,
                 from /home/giovanni/development/projects/mqtt-ir-control/mqtt-ir-control-01/mqtt-ir-control-01.ino:40:
/home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Udp.h:82:27: note: 	virtual IPAddress UDP::remoteIP() const
         virtual IPAddress remoteIP() const =0;
                           ^
/home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Udp.h:84:26: note: 	virtual uint16_t UDP::remotePort() const
         virtual uint16_t remotePort() const =0;
                          ^
Multiple libraries were found for "WiFiUdp.h"
 Used: /opt/arduino/libraries/WiFi
 Not used: /home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Do you have any suggestion about how to fix these errors?

Best regards,
Giovanni

pinootto, 2019-04-11:
Here are other logs with same error, when I try to compile IotWebConf01Minimal:

Arduino: 1.8.9 (Linux), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Build options changed, rebuilding all
In file included from /home/giovanni/Arduino/libraries/IotWebConf/examples/IotWebConf01Minimal/IotWebConf01Minimal.ino:28:0:
/home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/DNSServer/src/DNSServer.h:67:13: error: cannot declare field 'DNSServer::_udp' to be of abstract type 'WiFiUDP'
     WiFiUDP _udp;
             ^
In file included from /home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/DNSServer/src/DNSServer.h:3:0,
                 from /home/giovanni/Arduino/libraries/IotWebConf/examples/IotWebConf01Minimal/IotWebConf01Minimal.ino:28:
/opt/arduino/libraries/WiFi/src/WiFiUdp.h:27:7: note:   because the following virtual functions are pure within 'WiFiUDP':
 class WiFiUDP : public UDP {
       ^
In file included from /opt/arduino/libraries/WiFi/src/WiFiUdp.h:23:0,
                 from /home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/DNSServer/src/DNSServer.h:3,
                 from /home/giovanni/Arduino/libraries/IotWebConf/examples/IotWebConf01Minimal/IotWebConf01Minimal.ino:28:
/home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Udp.h:82:27: note: 	virtual IPAddress UDP::remoteIP() const
         virtual IPAddress remoteIP() const =0;
                           ^
/home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Udp.h:84:26: note: 	virtual uint16_t UDP::remotePort() const
         virtual uint16_t remotePort() const =0;
                          ^
Multiple libraries were found for "WiFiUdp.h"
 Used: /opt/arduino/libraries/WiFi
 Not used: /home/giovanni/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/ESP8266WiFi
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Best regards,
Giovanni

prampec, 2019-04-11:
Can you please try compiling some ESP8266 examples with this Arduino version?! I cannot do anything if the problem is in basic ESP8266 source code (what seams to be the case). We have to sort this out first. Further more you might want to help me with trying to compile for esp32 ("widora-air" hardware is recommended, as it has predefined pin mappings e.g. D1).

pinootto, 2019-04-12:
Hi prampec,

I compiled some ESP8266 examples with this Arduino version successfully.
Regarding esp32, I don't have esp32 hardware available now.

Best regards,
Giovanni

9H5G, 2019-06-16:
I have compiled examples on ESP32 without any problems.

closed 2021-04-14 by prampec