prampec/IotWebConf #40: Setting Hostname

electrontube, 2019-04-22:
Is there a way to set the the hostname?
Currently the device appears as: ESP_nnnnnn (when connected)
It would be nice to have an option to set hostName = thingName
or even better to have an extra field on the config page to set the hostname separately.

prampec, 2019-04-30:
IotWebConf tries to set the hostname to the name defined in the Web configuration page. However I also have faced with problems where it was not done. Not sure why.

electrontube, 2019-04-30:
Thanks for the response, which leads to the next question:
Will there be any further development soon?

prampec, 2019-04-30:
Regarding to this particular problem (or for any other problem), what I can say is, that when one can form a detailed reproducible report of an obvious error, than I will do my best correct that error.

Further more there is a "Feature request" section in the README, with the ideas I might want to address sooner or later.

But any contribution is very welcomed.

electrontube, 2019-04-30:
Thanks again for your response!
That sounds a bit like: "be patient, it will take quite a while"

prampec, 2019-04-30:
What can I say? :) You see it right.
The library is quite stable now, I have spent a lot of effort to achieve this level. Now there is nothing that hurries me.

all-solutions, 2019-05-08:
Try 
_void setup () {
[...]
WiFi.hostname(iotWebConf.getThingName());
[...]
}_

electrontube, 2019-05-12:
@all-solutions 
Thanks for your response, 

> 
> ```
> void setup () {
> [...]
> WiFi.hostname(iotWebConf.getThingName());
> [...]
> }
> ```

Did not make any difference, HostName still is ESP_nnnnnn

Any other ideas?

electrontube, 2019-05-12:
> 
> ```
> void setup () {
> [...]
> >  WiFi.begin();
> [...]
> }
> ```

seems to do the trick... 
can anyone confirm this?

closed 2019-07-09 by prampec