prampec/IotWebConf #41: Wrong config version

steinbeck, 2019-04-30:
I am getting the error
"Config size: 165
Wrong config version."
presumably, since I played with other WifiManagers before this one :)
Using   iotWebConf.setConfigPin(CONFIG_PIN); and pulling this pin to ground seems to have no effect. 
Is there a way to delete an "old" config?

prampec, 2019-04-30:
Right! So this is not an error message. It just notifies you in the console, that AP mode is forced. In this state, the AP mode will not be exited until you provide a (new) proper configuration in the web GUI. A proper configuration means, that you must change the initial password, and set the WiFi connection information of your local network.
(WiFiManager has nothing to do with this. :) )

closed 2019-04-30 by prampec

steinbeck, 2019-04-30:
Thanks! I realise that I did not notice this behaviour since I had an empty loop() method, so no iotWebConf.doLoop() and therefore never saw what that an AP was set up, etc. 
My attempted thingspeak temperature logger does everything in the setup method and then goes to deepsleep once successfully uploading things. I think I now understand how to do this with a non-blocking wifi manager such as iotWebConf. And I have to say that I like the implied better design. :)