prampec/IotWebConf #44: User-configurable library settings

EddieGreen, 2019-05-13:
Thanks for all the owrk on this library - really useful.

I'm already using EEPROM in setup()
`EEPROM.begin(512);`

Your library redefines it
`EEPROM.begin(IOTWEBCONF_CONFIG_START + IOTWEBCONF_CONFIG_VESION_LENGTH + size);`

With this, memory is messed up for your code and mine.

Having limited knowledge of C++, I cannot figure out how to specify my memory requirements without changing your library code, which seems to me to defeat the use of a library.

At the moment I have changed the following lines:
```c++
IotWebConf.cpp line 200 - //EEPROM.begin(IOTWEBCONF_CONFIG_START + IOTWEBCONF_CONFIG_VESION_LENGTH + size);

IotWebConf.h line 28 - #define IOTWEBCONF_CONFIG_START 336
```

Now I'm looking at other settings that can't be changed except by altering the library code, such as IOTWEBCONF_CONFIG_USE_MDNS, IOTWEBCONF_DEBUG_TO_SERIAL, and IOTWEBCONF_ADMIN_USER_NAME.

Am I missing something? Or can you make these settings and options configurable when initialising the library?

prampec, 2019-05-13:
No. You are right. These defines can only be set by changing the defines in the .h file. This is a kind of stupid way, but I don't see any other way to do this.
The idea behind the IOTWEBCONF_CONFIG_START is that let's say you have 336 bytes of user data you would like to store in the EEPROM. So you will store it from position 0 to 335, and have the IOTWEBCONF_CONFIG_START set up for 336.
Thus you do not have to exclude line 200, as the EEPROM.begin() will allocate just the right amount of memory for you.

EddieGreen, 2019-05-22:
Regarding EEPROM memory allocation, I tried this several ways while leaving IotWebConf.cpp line 200 untouched. None of them ended up writing the config or my data correctly to an ESP32. The permutations were:

My Program:
```
EEPROM. begin(512)
EEPROM. begin(336)
Not declared
```

IotWebConf (with IOTWEBCONF_CONFIG_START =336):
```
Line 200 in place
Line 200 commented out
```
The only way I got memory to play was with EEPROM.begin set in my program and commented out in your library.

Have I miscalculated the amount of space required for the default parameters? (4 x 32) + 4 = 132 bytes. If EEPROM. begin got a larger number than the memory available, would it start wrapping round to the start of the EEPROM area?

Regarding configuration in general, wouldn't it be possible to create a settings object that would be initialised with default but editable parameters, and passed to `iotWebConf.init()`?
 
```
iotWebConfSettings settings;
settings.memStart=336;
settings.statuspin=6;
settings.apTimeout=30;
settings.handleEEPROM=false;
iotWebConf.init(settings);
```

prampec, 2019-05-22:
Sorry, I do not know why the EEPROM.begin did not work out well for you when done within the library.

I don't think the index will wraps around, when larger value is provided. Instead some I would expect some misbehavior.

Using struct as a parameter is a good idea by itself. I will keep that in mind when creating/modifying interfaces in the future.

closed 2019-05-22 by prampec