prampec/IotWebConf #45: Crash when errorMessage is not initialized

rzeldent, 2019-05-19:
When rendering the page for the first time no error message is present for the thingName parameter. When replacing the {e} parameter in the template it crashes.

Solution: Initialize error message in constructor.

IotWebConfParameter::IotWebConfParameter(
  const char *label,
  const char *id,
  char *valueBuffer,
  int length,
  const char *type,
  const char *placeholder,
  const char *defaultValue,
  const char *customHtml,
  boolean visible)
{
  this->label = label;
  this->_id = id;
  this->valueBuffer = valueBuffer;
  this->_length = length;
  this->type = type;
  this->placeholder = placeholder;
  this->customHtml = customHtml;
  this->visible = visible;
  **this->errorMessage = "";**
}

Unfortunately could not make a branch/pull request. Could you check this and update the repo so will be fixed on PlatformIO?

Thanks!

prampec, 2019-05-20:
Thanks for the feedback! I will check your fix and issue a release ASAP.

rzeldent, 2019-05-21:
Hi Balazs,

That would be great. I looked around on PlatformIo and think you're the
only one that offers a decent solution for the ESP32.
I did make a provisioning library on my own but think it's better to reuse
and have a few but  well maintained libraries.

Looking forward to see the fixes, If you need any help let me know...

Kind regards

Rene Zeldenthuis

prampec, 2019-05-22:
Fix available from version 2.1.0.

closed 2019-05-22 by prampec

rzeldent, 2019-05-22:
Ho Balazs,

Thanks for fixing this!

Kind regards,

Rene Zeldenthuis

rzeldent, 2019-06-02:
Hello Balazs,

I saw the fix and included the library in the (public) esp32cam project.
Thanks for the library and fixing!

https://github.com/rzeldent/esp32cam-ready

Rene Zeldenthuis