prampec/IotWebConf #47: Q: WiFiScan

Webunity, 2019-06-04:
Can an addition be made that the list of found networks are displayed in AP mode?
Scan results can indicate if a wifi has protection, which could be used to hide the password field.

Configuration options i can think of are:
1) MinimumRssiNeeded
2) ListOpenNetworks(true|false)
3) ListSecuredNetworks(true|false)

prampec, 2019-06-05:
While this would be a very neat feature, I resist to add it. The alternative solution WiFiManager (as far as I see) suffers from this. Detecting other networks can be quite tricky and certainly can be very much dependent on the provided WiFi library (thus codes for ESP8266 will not work for ESP32).
Further more I would like to keep the code simple and with a moderate footprint. WiFi discovery is not likely to fit in this philosophy.
Instead, what I can imagine is a plugin feature, where one can extend IotWebConf to add extra features like this one. So my job here is to provide a possibility of extending the functionality.

Webunity, 2019-06-05:
I agree. It adds convenience but can be added by the developer themselves as well.

closed 2019-06-05 by Webunity

coelner, 2021-01-17:
What about WPS PBC (Push the button)?

prampec, 2021-01-18:
I was just thinking about the WiFi discovery addition the other day. At least we should put it to the wishlist. :)

The WPS is another thing. My main concern about that, is that is quite hard to test.