prampec/IotWebConf #53: Disable Serial Debug

Oj-Hond, 2019-06-26:
Hi thanks so much for the amazing library.

Is there a way to completely disable the serial debug without having to remove it from the library?

Thanks in advance

prampec, 2019-06-26:
It is certainly designed to be disabled with an easy step. However you need to modify the IotWebConf.h for that. (What is a pity, but we cannot do it better.)

So please exclude "#define IOTWEBCONF_DEBUG_TO_SERIAL" (line 43) in the IotWebConf.h for completely disable serial log!
(Disabling serial log for production environment is highly recommended.)

closed 2019-06-26 by prampec

Oj-Hond, 2019-06-26:
Hi @prampec perfect thank you so much, not a worry way better option than having to remove all of the serial prints.

geronet1, 2023-11-12:
It would be nice if to get debug output to Serial1 with a build flag:

IotWebConfSettings.h line 57:
`// -- Helper define for serial debug
#ifdef IOTWEBCONF_DEBUG_TO_SERIAL
#ifndef IOTWEBCONF_DEBUG_DEVICE
# define IOTWEBCONF_DEBUG_DEVICE Serial
#endif
# define IOTWEBCONF_DEBUG_LINE(MSG) IOTWEBCONF_DEBUG_DEVICE.println(MSG)
#else
# define IOTWEBCONF_DEBUG_LINE(MSG)
#endif
`

like this you can change it to Serial1:
`-DIOTWEBCONF_DEBUG_DEVICE=Serial1`

with this flag you can disable debug output:
`-DIOTWEBCONF_DEBUG_DISABLED`