prampec/IotWebConf #54: add svg graphics

fabltd, 2019-07-01:
Hi 

Any pointers on how to add an SVG to the config page?

prampec, 2019-07-01:
There are some request regarding extending the HTML capabilities of the config page. So this will be the next on my TODO list.

fabltd, 2019-07-01:
Thanks

Is there an ETA?

Regards,
Jeremy

prampec, 2019-07-04:
it is done by now. I'm need to test things tho. E.g. by provide an example

fabltd, 2019-07-04:
Hi

Is there an email I can send you a file to ?

J

prampec, 2019-07-04:
It is not ideal to include files in examples, as it needs to be uploaded to SPIFFS and used from that location. Makes the example very complicated and eventually impossible to run for beginners. Instead I'm thinking about using an inline PNG image as an example.

fabltd, 2019-07-04:
Hi

The SVG could be included in a header file ?

prampec, 2019-07-04:
What do you mean by that? What I will do is to provide options to customize the generated HTML code. I will not give advises on that how to manipulate HTML code. There are way to many online resources for that.

fabltd, 2019-07-09:
Can i test this now?

prampec, 2019-07-09:
The code and the example is ready on my computer, but unfortunately I have run into hardware/driver issues, after the first test: my computer does not recognize my ESP8266 any more. (I need to find another computer.)

prampec, 2019-07-09:
Done in v2.3.0

closed 2019-07-09 by prampec