prampec/IotWebConf #58: How to reset every config?

Geraldo-Neto, 2019-07-17:
Simple question: how do I reset every single config to its defaults ?

syrinxtech, 2019-07-21:
I've had to do this several times.  You just create a short routine to write 0's to the first X bytes of EEPROM, where X = size of the configuration.  If you're not sure, you could always write 1K worth, and that should work fine.

I'm looking for a way to do this to individual parameters instead of the whole EEPROM configuration.

mangtronix, 2020-06-05:
Here's a sketch that clears the whole EEPROM area. Useful for obliterating the stored config before reprogramming https://github.com/mangtronix/InternetOfOurThings/blob/master/eeprom_clear_esp/eeprom_clear_esp.ino

prampec, 2021-04-14:
Check #176, but basically you can e.g. use:
iotWebConf.getSystemParameterGroup()->applyDefaultValue();

Will be enabled with version v3.0.2

closed 2021-04-14 by prampec