prampec/IotWebConf #61: Default Values not working

jamez01, 2019-08-10:
IoTWebConf Version 2.3.0
Using an ESP8266 

```C#
IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
IotWebConfSeparator separator2 = IotWebConfSeparator("Nutrients");
IotWebConfParameter mlGallon = IotWebConfParameter("ML Per Gallon", "mlGallonParam", mlGallonValue, NUMBER_LEN, "number", NULL, "4.9", "step='0.1'");
IotWebConfParameter separator3 = IotWebConfSeparator("Advancied Config (Leave defaults if unsure)");
IotWebConfParameter ticksGal = IotWebConfParameter("Ticks Per Gallon", "ticksGalParam", ticksGalValue, NUMBER_LEN, "number", NULL, "3200", "min='1' max='100' step='1'");
IotWebConfParameter msMl = IotWebConfParameter("ML Per Second", "msMLParam", msMLValue, NUMBER_LEN, "number", NULL, "320.0", "step='0.1'");
```

The above configuration will show all custom parameters, but the set defaults are not in the input fields, nor does IoTWebConf set the appropriate defaults when the fields are left blank.

I know this issue was supposed to be fixed in 2.1.0, but seems to be reintroduced in 2.3.0.

prampec, 2019-08-12:
Thank you for your feedback. I'm just rewriting how IotWebConf handles the parameters, and I will also take care about this then.

prampec, 2019-08-12:
I still don't understand tho how the bug was reintroduced. The default values are used exclusively in IotWebConf::configLoad(), and I do not see any change in that lately.
I suppose, that it was never worked as expected. E.g. value buffers should contain a zero length string, so currently you might need to do that manually.
`mlGallon[0] = '\0';`

t0mg, 2020-10-28:
Hi there @prampec - I'm facing the same issue and I'm not sure where I'm supposed to apply your suggested workaround. It does not seem to have any effect. Could you please be more specific? 

Thanks!

prampec, 2021-04-14:
Default value handling is completely changed and hopefully clarified in v3.0.0. Please consult the corresponding section of the migration guide: https://github.com/prampec/IotWebConf/blob/master/doc/MigrationGuide-v3.0.0.md#default-value-handling

closed 2021-04-14 by prampec