prampec/IotWebConf #62: WiFi password length

newmindJP, 2019-08-20:
the input for WiFi password length is limited in the definition file to 33 chars -> for WPA2 the SSID can be max 32 char and the password 63 char

prampec, 2019-09-23:
Please be more specific! What is the issue here, and what is your opinion on solving the issue.
(Sorry for the late answer!)

eklee, 2019-10-13:
Maybe update this line to use a different char length of 64? this->_apPasswordParameter = IotWebConfParameter("AP password", "iwcApPassword", this->_apPassword, IOTWEBCONF_WORD_LEN, "password");

teaserrr, 2019-12-28:
Indeed the wifi password has a maximum length of 33 characters it seems. I found the limitation in IoTWebConf.h:
// -- Maximal length of any string used in IotWebConfig configuration (e.g.
// ssid, password).
#define IOTWEBCONF_WORD_LEN 33

Why is it limited? The password of my home wifi is longer than 33 chars so the library is unusable because of this.

4m1g0, 2020-01-01:
Just a comment about this: The maximum length of the WPA passphrases is 63 characters in order to distinguish them from the actual PSK which is 64 ASCII characters long. If you provide a 64 chars string, it is assumed to be a PSK, otherwise is treated as a passphrase and is used to compute the PSK.

So the char length should be 65 to allow for the 64 allowed characters plus the null terminator.

https://github.com/esp8266/Arduino/pull/1850

prampec, 2021-04-14:
You are perfectly write. It was a mistake not to use the standards here.
But as mentioned in #166, we are already too far to fix this.
However you still have the option to change password length on your side, please consult documentation for compile time configurations: https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#compile-time-configuration

closed 2021-04-14 by prampec