prampec/IotWebConf #63: How to work only in AP mode

manuprendlair, 2019-08-22:
Hello,

This not an issue but more a question:
In my use I just need that my esp32 work in AP mode. But I would like to be able to change the SSID or password of the AP, and other specific things. (this AP will broadcast data by udp)

So how to make IotWebConf only working in AP mode (and not switch to station mode and try to connect to a network.) ?

Thanks

Emmanuel.

prampec, 2019-09-23:
You might want to mess around with setWifiConnectionHandler() to define a custom wifi connection method, that doesn't actually connect to the WiFi network. But I do not think this will does the job for you. (Sorry for the late answer.)

excelly2k1, 2019-10-12:
I am also in this same situation where i don't want my ESP8266 try to connect to the wifi network as there is no wifi network available. But ESP8266 keep on changing its state from ap to station mode and try to connect to the wifi network which i used at the initial setup in my location and was now in another location intended for the actual use.

shmick, 2020-02-10:
I was wondering if any progress has been made with this? I've also got a project that doesn't need the unit to connect to WiFi, but I need to connect to the AP at random times in order to tweak some custom parameters.

excelly2k1, 2020-02-20:
**This will indeed save the esp power consumption.** & also improve the loop speed.

prampec, 2021-04-14:
There is a method introduced: forceApMode(), when you call it with TRUE, it will enter AP mode and will not release it. When called with FALSE, the locked AP mode is released.
There is also a feature request (#149), to completely stop WiFi network on request. This is very likely be implemented some times.

closed 2021-04-14 by prampec