prampec/IotWebConf #64: Most of times not shows the configuration page

alexjalser94, 2019-09-02:
Hi! I´m using this in ESP32, but I have an issue that I can´t resolve, the only way to view the configuration page is when I connect to the AP and immediately reload a page in the browser with 192.168.4.1/config, but if only I connect and wait till this page it doesn't show...
I don't know what can occur, but I only can run it correctly if do the trick that I mentioned previously, can you help me? Maybe if we can force the configuration page by the .ccp can resolve this, but I have no idea where I have to touch...

This one I when I startup normally and I connect to the AP, but it stuck as you can see

[image]

And this one (showing that is runs) is when I am reloading in the browser during the connection to the AP, and changing some values

[image]

Please help me to solve it! Thanks :)

balajtii, 2019-09-04:
Hi! Same here. After I stuck with error repair, I opened a new project and copy/paste my old code block by block (logical blocks eg connection management, mqtt connection, mqtt topic name generation...etc) and testing wich block give the error and nothing, my new file works great with almost the same code. I don't get the solution.

prampec, 2019-09-23:
Sorry I cannot provide any hints for you on that, expect for encourage you to experiment using different ESP framework versions.
If you have any solutions since the problem occurred, I would like to share it with others. (Or fix the problem if there is any.)

prampec, 2021-04-14:
There was some stability issues fixed with v3.0.0 . All what we can hope is that is also fixes these problems. :)

closed 2021-04-14 by prampec