prampec/IotWebConf #65: Change Button Option

fabltd, 2019-09-03:
Hi 

Any way to have the portal on startup only load if the button is pressed?

Thanks

J

MichGraf, 2019-09-10:
Try „iotWebConf.skipApStartup();“ before init.
Attention: the Update-Server does not work in this case. See my following issue.

closed 2019-09-23 by prampec

bedawi, 2020-12-23:
I also had the problem that the access point mode with captive portal was loaded every time at reboot - even with a configured WiFi connection. I figured out that I had missed to configure the pin as Input. Not all examples have this line and therefore its easy to miss:

```
pinMode(CONFIG_PIN, INPUT_PULLUP);
```

You can, for example, set this up in the setup() function together with skipApStartup():

```
pinMode(CONFIG_PIN, INPUT_PULLUP);
iotWebConf.setConfigPin(CONFIG_PIN);
iotWebConf.skipApStartup();
```