prampec/IotWebConf #68: IotWebConfParameter checkbox type

SheepKnight, 2019-09-07:
Whenever a IotWebConfParameter is defined with parameter type set as "checkbox", no results are saved.

prampec, 2019-09-23:
Yes. Checkboxes, radio boxes, and special HTML field like that might be a bit tricky to handle. I have already started to rewrite input field rendering in IotWebConf, to provide a much more convenient way to do this, but I did not have the time lately to finish the tests.
So you might receive so changes on that sooner or later.

t0mg, 2020-10-28:
I'm using a range input type (with `min=0 max=1 step=1`), to mimic checkboxes. It works out of the box with the current version of the lib.

```ino
IotWebConfParameter("Int as toggle param", "intParam", intParamValue, NUMBER_LEN, "range", "0", "0",
    "data-labels='Off|On' min='0' max='1' step='1'");
```

For more convenience, I've also added custom JavaScript (following [this example](https://github.com/prampec/IotWebConf/blob/master/examples/IotWebConf10CustomHtml/IotWebConf10CustomHtml.ino)) to 
- display the current value as a label next to the range input
- optionally override these labels (eg Off, On instead of 0,1).
```js
// Adds a label next to Range inputs
document.querySelectorAll("input[type=range]").forEach((inputEl) => {
    const labels = inputEl.getAttribute("data-labels");
    const labelList = labels && labels.split("|");
    const setLabel = () => {
        inputEl.setAttribute("data-label", labelList ? labelList[parseInt(inputEl.value, 10)] || inputEl.value : inputEl.value);
    };
    inputEl.oninput = setLabel;
    setLabel();
});
```
and some custom CSS to position things correctly
```css
input[type="range"] {
  margin-left: 30px;
  position: relative;
  padding: 0;
  width: 30px;
}
input[type="range"]:before {
  content: attr(data-label);
  color: #eee;
  position: absolute;
  left: -30px;
  display: inline;
  width: 25px;
  line-height: 20px;
}
``` 

Note that with this implementation you can also provide `data-labels='Off'` in a wider range input, to only replace the value 0 with a custom label (I use it for a sensitivity level setting).

prampec, 2021-04-14:
With v3.0.0 checkbox is already available (with many other exciting new features).

closed 2021-04-14 by prampec