prampec/IotWebConf #69: add setThingName

WillTemeraire, 2019-09-08:
Hello,
please add a function to dynamically change the variable "ThingName" (=SSID) in the code before calling iotWebConf.init();.

The possibilities to assign the ESP32 (e.g. the ChipID) are not solved properly (please compare Issue #11 )

Thank you very much, sir.

prampec, 2019-09-23:
I'm not sure in the moment. But the getThingNameParameter() will provide you full control on the variable. Isn't this what you are looking for?

mmmstew, 2019-11-04:
WillTemeraire means that currently the only way to provide SSID is statically through constructor. This is a problem if main loop or setup code needs to generate SSID based on chip id, random number, time etc. One way is to modify the library as shown in #11. I was able to do it with custom connection handler but it is a bit ugly:

Define global iotWebConf with temporary static ssid, password:
`IotWebConf iotWebConf("temp_ssid", &dnsServer, &server, "temp_pwd");`

Create custom connection handler before iotWebConf.init();
`iotWebConf.setApConnectionHandler(&connectAp);`
`iotWebConf.init();`

Custom connection handler:
`boolean connectAp(const char* apName, const char* pwd) {`
`  // ignore function parameters, generate your own dynamically generated ssid, password:`
`  return WiFi.softAP(ssid, password, 4);  // 4= max number connections`
`}`

PS I tried to first create global pointer (IotWebConf *iotWebConf) and placed constructor in setup code with dynamic SSID. This worked except then it would unexplainably crash and reboot the esp32 core when captive portal web page /config was accessed.

fermuch, 2019-12-01:
I've had the same problem (dynamic SSID) and this is my temporary solution:

```cpp
String dynamicThingName = "foobar";
auto iotThingName = iotWebConf.getThingName();
strncpy(iotThingName, dynamicThingName.c_str(), IOTWEBCONF_WORD_LEN);

iotWebConf.init();
```

ThomasWaldmann, 2020-01-12:
@prampec it would be nice to have an official setThingName complementing getThingName to avoid hacks like in previous post (which I will use for now).

In our project https://github.com/ecocurious/MultiGeiger there was even a full copy of your library source, edited just to add setThingName...

richiejarvis, 2020-02-09:
+1 for this feature too!

ThomasWaldmann, 2020-02-09:
@richiejarvis in the end, we solved it by creating the class instance right with the correct thingName, so we neither needed the dirty hack nor a not-yet-implemented setThingName.

prampec, 2021-04-14:
This is an old one. Can someone please summarize whether there is a feature request here, or not. Thanks!

EricDuminil, 2021-05-17:
Isn't it possible to declare a global pointer to iotWebConf:
`IotWebConf *iotWebConf;`
and create the actual object later (see this [SO answer](https://stackoverflow.com/a/7557195/6419007)), when more logic can be used in order to set ThingName?

```
setup{
    iotWebConf = new IotWebConf(getThingNameFromSomewhere(), &dnsServer, &http, http_password, config_version);
    ...
}
```

You'd just need to call the methods with `->` and not `.`, e.g. `iotWebConf->addParameterGroup(&csvParams);`

prampec, 2021-05-17:
@EricDuminil I would avoid using the "new" keyword in a microcontroller without a proper operating system. You might end up with memory problems in random points of the execution. In such a case you will not be notified about the problem, the MCU will just hang until the watchdog resets it.

EricDuminil, 2021-05-17:
@prampec Thanks for the feedback (and the very good library). If `new` is used once, at startup, for a global object, could it still lead to memory problems afterwards?

EricDuminil, 2022-02-23:
@prampec : It's true that "new" can be dangerous on a microcontroller. As far as I can tell, `new IotWebConf()` only takes 4 bytes, though, once all the parameters have been defined anyway. I can live with that, and it makes it easy to define thingname, SSID and password dynamically.

From what I understand, IotWebConf uses many Arduino `Strings`, which all call `malloc` in the background. Using them, replacing them and concatenating them call malloc and realloc multiple times, and leave the heap fragmented.

Did you ever consider switching to c-strings?

prampec, 2022-03-03:
@EricDuminil You are most probably right. Unfortunately I'm not well experienced in C++, I just try to survive with my C knowledge.

EricDuminil, 2022-03-03:
@prampec No need to be too modest : you wrote a great library!

C knowledge is exactly what you'd need if you ever want to replace Arduino Strings with char arrays.
https://cpp4arduino.com/2020/02/07/how-to-format-strings-without-the-string-class.html and https://hackingmajenkoblog.wordpress.com/2016/02/04/the-evils-of-arduino-strings/ helped me transition to char arrays for my projects.

Without changing much code, it might be a good idea to at least reserve (https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/reserve/) memory for the String objects, in order to reduce heap fragmentation.