prampec/IotWebConf #70: IotWebConf.delay breaks on ESP32

MichGraf, 2019-09-09:
I‘ve read that a delay(1) does not work at the ESP32. It restarts.
A change in the .cpp file from  delay(1) to delayMicroseconds(1000) works fine.

prampec, 2019-09-23:
Do you know the reason why delayMicroseconds() works and delay() don't? Probably again because of that stupid yield().
(I personally do not use the lib on ESP32 for daily basis.)

MichGraf, 2019-09-25:
Hi

Look at this Board:

https://github.com/espressif/arduino-esp32/issues/903

Regards 
Michael

prampec, 2019-09-25:
Thank you for referencing that. Okay I will fix it some days, if it also works with ESP8266.

1heinz, 2019-10-10:
I can confirm, that on esp32 almost all connections to a wifi network are given up.
With the delayMicroseconds(1000) instead of delay(1) all connections work instantly.
Why not use ?

```
#ifdef ESP8266
delay(1)
#else
delayMicroseconds(1000)
#endif
```

prampec, 2021-04-14:
Fixed.

closed 2021-04-14 by prampec