prampec/IotWebConf #71: Update-Server no Setup when skipApStsrtup

MichGraf, 2019-09-10:
A wonderful library😊😊👍👍

One Issue: if i run with skipApStartup the Link to update the firmware is shown. But a click on it: „File Not found“.
I saw in the .cpp file that a Update-server-setup is only be done while in AP-Mode or when not configured.
Add the setup in the State „_state_online“ before „_updateCredentials“ and it works for me. I‘m sure this is not a clean solution.
A question: Connecting to WiFi works often not at the First time. Maybe 50:50. Can you imagine a reason?

prampec, 2019-09-23:
No, you are not right. Update-server-setup also done in STATE_ONLINE, but when in AP (and in not-configured) the update-server does not need password, in STATE_ONLINE it does.
(No, I'm not sure about why connecting fails. But try to do less thing in loop(), it might help.)

MichGraf, 2019-09-25:
Thanks for the answer.
?? In my environment everytime i get a „page not found“ trying to do an update if i use  skipApStartup?? With an additional setup like in my pull-request it works stable for me. I don’t use a password in the moment.

Feel free to cancel the pull request. It was meant  as hint.

Regards
Michael

prampec, 2019-09-25:
Okay. Thank you for the pull request. Now I know exactly what you mean. I have some time to investigate that tho. (I will drop your pull request, as I want to have to separate commits for the two fixes.)

prampec, 2020-01-14:
Introduced skipApStartup. Will be available from version v2.3.1.

closed 2020-01-14 by prampec