prampec/IotWebConf #81: IotWebConf with Blynk combiantion is very unstable with iotWebConf.init()

Msprg, 2019-11-01:
I am not really sure wether is this IotWebConf, Blynk, or ESP Core´s fault, but here it goes...

Blynk is a pretty good and simple to use IOT platform. I needed to manage wifi conection - got IotWebConf for that, and Blynk for the "IOT" part of my project.

MCU: ESP8266 as WeMos d1 mini.

The issue is that it looks like the blynk is unable to make stable connection to the server. The issue only appears if the IotWebConf was initialized. If IotWebConf was not initialized, the blynk seems to connect just fine (after the network was set-up of course). Does anyone know what can be causing this behaviour?

Here is the code (minimal example merged with blynk),
The code should be runnable as it is, if you have an ESP8266, libraries, and Internet connection:

```
#define BLYNK_PRINT Serial

#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

#include <IotWebConf.h>
char auth[] = "G4aJD5mCG-mJ-w-OLlZkTZmns760JwC0";

// -- Initial name of the Thing. Used e.g. as SSID of the own Access Point.
const char thingName[] = "testThing";

// -- Initial password to connect to the Thing, when it creates an own Access Point.
const char wifiInitialApPassword[] = "smrtTHNG8266";

DNSServer dnsServer;
WebServer server(80);

IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword);

void setup()
{
  Serial.begin(74880);
  Serial.println();
  Serial.println("Starting up...");

  // -- Initializing the configuration.
  iotWebConf.init(); // Try to comment out this init, and see wether the issues are gone...

  // -- Set up required URL handlers on the web server.
  server.on("/", handleRoot);
  server.on("/config", [] { iotWebConf.handleConfig(); });
  server.onNotFound([]() {
    iotWebConf.handleNotFound();
  });

  Blynk.config(auth);

  Serial.println("Ready.");
}

int state;
void loop()
{
  iotWebConf.setApTimeoutMs(2665);
  // -- doLoop should be called as frequently as possible.
  iotWebConf.doLoop();
  Blynk.run();
}

/**
   Handle web requests to "/" path.
*/
void handleRoot()
{
  // -- Let IotWebConf test and handle captive portal requests.
  if (iotWebConf.handleCaptivePortal())
  {
    // -- Captive portal request were already served.
    return;
  }
  String s = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/>";
  s += "<title>IotWebConf 01 Minimal</title></head><body>Hello world!";
  s += "Go to <a href='config'>configure page</a> to change settings.";
  s += "</body></html>\n";

  server.send(200, "text/html", s);
}

```

prampec, 2019-11-01:
As far as I see, Blynk manages WiFi configuration by itself. So it might happen, that both IotWebConf and Blynk tries to connect/disconnect from networks parallel.

Msprg, 2019-11-01:
> As far as I see, Blynk manages WiFi configuration by itself. So it might happen, that both IotWebConf and Blynk tries to connect/disconnect from networks parallel.

I think not the case, because I intentionally did not provide network SSID and passphrase for blynk. Also AFAIK Blynk just does `WiFi.begin(ssid, passwd);` when `Blynk.begin(auth, ssid, passwd);` is called which is not called in my code on purpose. I just called `Blynk.config(auth);` which is just telling the Blynk the auth token of the device...
I will try to get blynk to investigate a bit...

EddieGreen, 2021-03-06:
I'm experiencing this with ESP8266 devices, but not with ESP32s as far as I can see.

I use the onboard led for status. When it hangs, the LED is solid on. I've attached a serial port to figure out what's going on, which provides no indication of a crash.

I'm thinking along the same lines, that Blynk and IotWebConf are interfering with each other.

Can you try making `Blynk.run()` conditional, ie:
```
void setup()
{
    Blynk.config(.....) // do not start with Blynk.Connect() anywhere
    ...
}

void loop()
{
    if(IotWebConf.getState() == IOTWEBCONF_STATE_ONLINE)
    {
        Blynk.run(); // will "Blynk.connect()" to WiFi automatically if it's not already connected - stop it from doing this automatically
    }
    ...
}
```

EddieGreen, 2021-03-06:
I should have also mentioned that, with around 12 devices in use, its the ESP8266's that are connected to the less reliable (more clients, higher traffic) access points that are performing this way.

Either way, I get the impression that once Blynk can't get what data it wants it returns to a connect cycle or otherwise interferes with the Wi-Fi connection.

It would be good to get more info on this process.

Msprg, 2021-03-14:
This issue is stalled, and as I've moved on, I just resorted not to use Blynk and IotWebConf together, also it would appear that no one else is having these issues, except @EddieGreen - that is one person per almost one and a half year... So I think I'll close this for now, until somebody in future will have similar problems, or it might even get fixed...

closed 2021-03-14 by Msprg