prampec/IotWebConf #82: setApTimeoutMs not working

mmmstew, 2019-11-04:
Using an ESP32, I try to set AP timeout using setApTimeoutMs(...) before running init(), but the value doesn't take effect.

Upon power up the device establishes a client connection to my WiFi router with existing settings after about 25 seconds during which time the captive portal AP is available. It gets disabled as soon the client connection is established.

I looked through the library and found:

1) IotWebConf::setApTimeoutMs() does not work because IotWebConf::init() overrides _apTimeoutMs with the integer conversion of _apTimeoutStr.

2) _apTimeoutStr is never initialized.

3) The NVM/EEPROM parameter _apTimeoutParameter is never used.

mmmstew, 2019-11-05:
I can confirm after removing all references to _apTimeoutStr and _apTimeoutParameter and setting _apTimeoutMs to IOTWEBCONF_DEFAULT_AP_MODE_TIMEOUT_MS in the class constructor, I can now use setApTimeoutMs() to set AP timeout correctly . Of course it is not saved in EEPROM but I don't need this in my application.

vik-patel, 2020-12-23:
I'm seeing the same issue with ESP8266 ... I'm using 3.0-Master ... the issue is if:

iotWebConf.getApTimeoutParameter()->visible = false;

any config changes results in iwcApTimeout/ApTimeoutParameter being set to '' in eeprom ... and any reboots when configured SSID is not available the AP is only visible for 1sec (the minimum).

How do we set the default of ApTimeoutParameter? 

Thx

vik-patel, 2020-12-23:
@prampec please can you help here ... thx

prampec, 2020-12-27:
I have fixed this yesterday. Please check!

vik-patel, 2020-12-28:
yes ... thx! @prampec seems to work now

prampec, 2020-12-28:
Thanks for the feedback!

closed 2020-12-28 by prampec