prampec/IotWebConf #83: WiFi will not connect until config portal timeout (ESP32)

mmmstew, 2019-11-05:
I am using ESP32 and noticed that on start-up it will not connect as client to existing valid WiFi network until the config portal AP times out. This is not related to my previous mention of AP timeout bug (#82 ).

After some investigation I added the following lines directly under `if (this->_state == IOTWEBCONF_STATE_BOOT) {` in the member function `IotWebConf::doLoop()`:
`WiFi.mode(WIFI_AP_STA);`
`this->_wifiConnectionStart = millis();`
`this->_wifiConnectionHandler(this->_wifiAuthInfo.ssid, this->_wifiAuthInfo.password);`
	
Also I changed `WiFi.mode(WIFI_AP)` to `WiFi.mode(WIFI_AP_STA)` in `IotWebConf::setupAp()`

This seems to work but not sure if this is the right way to fix?

WolfgangFranke, 2020-01-13:
I would like to know if there is an (official?) way to let the ESP32 on Power-On directly connect a valid WiFi network instead of waiting 30s until the AP times out?
For my device it would be good that it immediatley after POR or re-boot connects to a known WiFi.
Any thoughts are welcome.

prampec, 2020-01-13:
There is a skipApStartup() option, where you can force the device to start directly in station (WiFi) mode. (If WiFi connection fails, the AP will be still entered after all.)
However mixed mode (WIFI_AP_STA) is not supported, as it does not make too much sense. (I agree, that e.g. limit the uploading of a new firmware for AP mode would be a great security option, that certainly would make sense.)

closed 2020-01-13 by prampec

WolfgangFranke, 2020-01-14:
thank you, works like a charm ;-)