prampec/IotWebConf #84: How to disable Captive portal redirection

rick-pit, 2019-11-09:
Hi, can you help to disable redirections to the Captive portal like wifimanager (development version) does.

WifiManager method
`//if false, disable captive portal redirection
    void          setCaptivePortalEnable(boolean enabled);`

prampec, 2020-01-13:
What is the point of disabling the captive portal redirection?

TheKelmore, 2020-01-30:
For Security reasons i think, if you are creating a thing that your are putting in a company's network, or in a house, we could disabling the captive portal. I know that is needed the user and the password, but if someone intercepts this data, the thing will be vulnerable.

prampec, 2020-01-30:
Captive portal comes up only when the Thing is in AP mode and one was connected to the network provided by the Thing. That cannot be done without knowing the password of the network. As far as I know currently there is no way to grab network traffic without having connection to that network. So AP mode is the most secure way to configure your Thing, where captive portal is now the default.

What I can imagine tho, is that AP mode is the regular use of the Thing, e.g. with published password. Now, IotWebConf is not designed to handle this scenario.