prampec/IotWebConf #86: Examples of how to retrieve custom parameters

crumbliness, 2019-11-14:
Are there any examples of how to retrieve/get custom string parameters. I am unable to figure it out from the examples provided. I have the custom parameter example working and merged with an existing sketch.

feder-ico, 2019-11-14:
What do you mean with retrieve/get parameters?

With this as an example:
`IotWebConfParameter stringParam = IotWebConfParameter("String param", "stringParam", stringParamValue, STRING_LEN);`

you can use stringParamValue as char. For instance, you can do: `serial.print ((String) stringParamValue);` 

Is that what you need?

If you change the value of that variable and you want it to be persistant you will need to do `iotWebConf.configSave();` to save the value on EEPROM but bear in mind that configSave() will make the board to reboot.

Hope this helps.

crumbliness, 2019-11-15:
OK thanks, - that works. My problem then is how I am using the
stringParamValue(s) in an http: string. . I have a "invalid operands of
types 'const char*' and 'const char [26]' to binary 'operator+'" error.
 I'll need to study some more and try to sort that out.

Thanks again.

closed 2020-01-05 by prampec