prampec/IotWebConf #90: Crashes when used with ArduinoMenu

dboon1978, 2019-12-12:
Trying to use this menu system ( https://github.com/neu-rah/ArduinoMenu ) with iotwebconf and it crashes on boot every time.  

Serial error reads:
Config size: 165
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x400d9034  PS      : 0x00060034  A0      : 0x8008180c  A1      : 0x3ffbe790  
A2      : 0x3ffbef30  A3      : 0x20000000  A4      : 0x00000400  A5      : 0x3ffb1c40  
A6      : 0x00000026  A7      : 0x3ffc8180  A8      : 0x800810a4  A9      : 0x3ffb8060  
A10     : 0x3ffc173c  A11     : 0xcfc00000  A12     : 0xfffffff7  A13     : 0x3ffb1c20  
A14     : 0x00000020  A15     : 0xffffffff  SAR     : 0x00000019  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40001609  LEND    : 0x4000160d  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40087ad7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400d9034

Backtrace: 0x400d9034:0x3ffbe790 0x40081809:0x3ffbe7b0 0x400850ad:0x3ffbe7d0 0x40087ad4:0x3ffb1c80 0x400fd031:0x3ffb1cf0 0x400fb8f9:0x3ffb1d10 0x400fbd47:0x3ffb1⸮

prampec, 2019-12-13:
As far as I see, this is an ESP32 error dump, but I cannot see the stack trace, so it is not trivial for me to find out the specific problem here.
What I can recommend is to try the latest master branch of the IotWebConf, as the very latest commit might fixes a potential ESP32 problem, and that fix is not released yet.
Also, there is commit approaching soon, that might also fixes some issues, I try to speed up this one.

dboon1978, 2019-12-13:
Thanks Prampec!  So I'll try that today.  I did try to comment out some lines to see if I could identify how to fix the problem and the two lines that are causing the crash are:
1) during setup: IotWebConf.cpp line 200 EEPROM.Begin(...)
2) during loop: IotWebConf.cpp line 879 (setupAP) and line 884 (this->_server->begin()

I'll try the latest commit though.  Hopefully that'll work.   In which case i'll close the issue.

dboon1978, 2019-12-13:
OK tried the newest copy of the project and still crashes.  Same "fixes" still prevent crashing.  I'll try to break them down farther.

prampec, 2019-12-13:
Thanks for the investigation you've done already! Unfortunately I still do not see why it is failing.

dboon1978, 2019-12-14:
if I can find more i'll let you know.  thanks for the responsiveness.  have to work on other features but i'll come back to this.