prampec/IotWebConf #96: support multiple WiFi connections (feature request)

WolfgangFranke, 2020-01-12:
Some IoT things needs to work in different places with different WiFi. Those kind of use cases mean the need for an option to configure multiple WiFi connections, as you already described, with automatically trying the next WiFi settings when the last used one is not available.

WolfgangFranke, 2020-01-12:
Dear Balázs and contributors,
thanks for the IotWebConf library. Today, I chose to work with it in my ESP32 Solar Energy Meter (a project using an INA219 to visualize the energy harvest from small solar cells, to power-up some IoT ZigBee things). My final goal is to send the measured solar energy data to an IoT Cloud for visualization.
The IotWebConf library works super and fully automatic, I like that simplified usage over other WiFi Managers with too many options and dependencies.

WolfgangFranke, 2020-01-12:
There is no need to have the user managing and selecting the WiFi settings themself manually, nor to develop expensive UIs.
I believe the WiFi selection should be fully automated and silently integrated in the current IotWebConf library.

I would suggest to store 3 WiFi configs in EEPROM, and have the library do an automated flow like this:
- try to connect to the "last recently used" WiFi, if success then done.
- if that is not available, then try the second WiFi config, if success then flag as "last recently used", and done.
- if second WiFi is not available too, then try third WiFi config, if success then flag as "last recently used", and done.
- if all 3 WiFi settings did not connect, then open Access Point and ask for the new WiFi settings.
- important: to not disappoint users by deleting the "last recently used" WiFi config, which is probably the most important to them, it's better to overwrite the oldest unused config with the new WiFi config from AP.
- I personally would set the timeout for each step to 10s or 15 s, as I see all my ESP32 connecting very fast (<2s).

WolfgangFranke, 2020-01-12:
As an initial idea I would suggest the following new EEPROM structure which contains 3 WiFi configs, and also works as the LRU list to determine the WiFi to connect after Power-On, as well as the WiFi config to be deleted when a new WiFi config is entered in AP:

BYTE- USAGE of EEPROM
000 - special use for back-level compatibility (1 byte)
001 - WiFi-3 key "configVersion" (4 byte)
005 - WiFi-3 SSID (33 byte, string for max lenght of WiFi SSID)
038 - WiFi-3 Password (65 byte, string for max lenght of password)
103 - WiFi-2 key "configVersion" (4 byte)
107 - WiFi-2 SSID (33 byte, string for max lenght of WiFi SSID)
140 - WiFi-2 Password (65 byte, string for max lenght of WiFi password)
205 - WiFi-1 key "configVersion" (4 byte)
209 - begin of original structure, also containing WiFi-1 SSID&Password
xxx - end of data structure (it allows to variable lenght of stored IoTWebConf variables)

WolfgangFranke, 2020-01-12:
No user interaction will be necessary to manage the 3 WiFi configs, by the following implementation idea:
- try to connect to the WiFi-1 (config @EEPROM 205), if success then proceed as normal.
- if that first is not available, then try WiFi-2 (config @EEPROM 103) ... if success then copy EEPROM content of WiFi-1 to WiFi-2, and save the current WiFi data (was WiFi-2) to WiFi-1, that maintains LRU list, then proceed as normal.
- if second WiFi is not available either, then try WiFi-3 (config @EEPROM 001) ... if success then copy EEPROM content of WiFi-2 to WiFi-3, and copy EEPROM content of WiFi-1 to WiFi-2, and save the current WiFi data (was WiFi-3) to WiFi-1, that maintains LRU list, then proceed as normal.
- if all 3 WiFi settings did not connect, then open an Access Point and ask for the new WiFi settings. If the new SSID is the same as the stored WiFi-1 SSID, then save the new password as WiFi-1 password (just a password change, not changing LRU list). If the new SSID is different from the WiFi-1 SSID, then copy EEPROM content of WiFi-2 to WiFi-3, and copy EEPROM content of WiFi-1 to WiFi-2, and save the current WiFi data (from AP) to WiFi-1, that maintains LRU list, then proceed as normal. The LRU list makes sure to always delet the oldest config (WiFi-3 in LRU list).
- to find out if WiFi-2 and 3 are valid, the key "configVersion" could be checked. Or simply write a NULL at the last byte of each variable in EEPROM to terminate the string, then the connection just fail and the library works as normal.
- reading some specs, an EEPROM can do several thousand write operations without damage, that means even a daily write would be no issue for several years.

WolfgangFranke, 2020-01-12:
Optional: Backward compatibility:
If you have a lot of existing customers, you can automatically copy their existing EEPROM data into the new structure by:
- check e.g. in function "IotWebConf::init", if the old address of key "configVersion" contains a valid key (today stored: IOTWEBCONF_CONFIG_START = 0..3), 
- if the key is found on the old place, then copy the data (not sure if "IotWebConfParameter" or "valueBuffer" is stored in EEPROM?) to the new EEPROM address range, indicated in the suggested EEPROM structure above.
- we better copy the data by starting from the end of the address range and finish with the lowest byte, to prevent overwriting in case of overlapping address ranges.
- then write to EEPROM address 000 a byte that destroys the key there,
- proceed with the new code as normal, the new library code will now find the new data structure in EEPROM.
- all the next boots will go straight, as it does not find the old key anymore on address 000.

prampec, 2020-01-13:
(This issue got a little bit too long. I even need to find time to read it all.)
But in short: **Supporting multiply WiFi connection settings is definitely something I want to do**.
See method (setWifiConnectionFailedHandler()). Also refer to feature branch: CustomizableParameter

d-a-v, 2020-12-09:
>  But in short: Supporting multiply WiFi connection settings is definitely something I want to do.

Any update on this ? Would `ESP8266WiFiMulti` be of any help ?
That, with the available AP list on captive portal, would be great !

prampec, 2020-12-09:
I'm currently heavily working on branch CustomizableParameter, that will soon the Dev branch, and eventually the new v2.0.0 release very soon. (In a month.)
If that is done, then we have the toolkit to make multiply WiFi connections in terms of trying to connect to a next network is the first, second... is not available.

d-a-v, 2020-12-10:
Thanks for your feedback !

prampec, 2020-12-22:
v3.0.0 is out in the git repository. You can also find a HackingGuide covering this very issue.

prampec, 2021-01-14:
Multiply WiFi connection is supported from now on. You can find documentation on this in the [HackingGuide](doc/HackingGuide.md). It is available in the master git repository, will be published as a release in v3.0.0.

closed 2021-01-14 by prampec