prampec/IotWebConf #97: How to change Timeout values and Debug levels ?

Greece2, 2020-01-13:
Hi, I'm pretty new to programming, so the answer might be obvious to others ;-)
How can I change the timeout values (AP timeout, Client timeout without making
changes directly in the library ?
Adding `iotWebConf.setApTimeoutMs(60000);`
here
`  // -- Initializing the configuration.
  iotWebConf.setStatusPin(STATUS_PIN);
  iotWebConf.setConfigPin(CONFIG_PIN);
  iotWebConf.setApTimeoutMs(60000);
  iotWebConf.init();`
doesn't seem to make any change.

Same goes for the different Debug levels, like
`IOTWEBCONF_DEBUG_PWD_TO_SERIAL ?`
How and where to place it in the code ?

Thanks for supporting a newcomer !

prampec, 2020-01-13:
You are perfectly right on that this should be obvious to do. However in Arduino environment
unfortunately there is no proper way to do it.
What you can do, is to modify these values in your library/IotWebConf/src folder, but keep in mind, that these changes are temporary, and will be overwritten on next version update.

You have much more options with PlatformIO environment, but IotWebConf currently does not support these options. This is why I keep this issue open, until it will be fixed.

Greece2, 2020-01-14:
Okay, thanks for the information.
Still, I can't see why setting the AP-Time isn't possible without chaning the lib itself.
In the lib, there is a function to set the timeout:
`  /**
   * This method can be used to set the AP timeout directly without modifying the apTimeoutParameter.
   * Note, that apTimeoutMs value will be reset to the value of apTimeoutParameter on init and on config save.
   */
  void setApTimeoutMs(unsigned long apTimeoutMs)
  {
    this->_apTimeoutMs = apTimeoutMs;
  };`
Still, `iotWebConf.setApTimeoutMs(60000);` doesn't seem to do anything.

Same goes for:
`    /**
   * IotWebConf tries to connect to the local network for an amount of time before falling back to AP mode.
   * The default amount can be updated with this setter.
   * Should be called before init()!
   */
  iotWebConf.setWifiConnectionTimeoutMs(50000);`
seems to have no effect.

prampec, 2020-01-14:
Here you are! So the timeout you can set from the code.
What I was talking about is the compile-time options (defines). E.g. debug level, word length, etc.

prampec, 2020-01-14:
Okay. Fixed the defines, so that they can be altered externally build time. Will be available from version v2.3.1.

closed 2020-01-14 by prampec