prampec/IotWebConf #98: remove hardcoded strings

ThomasWaldmann, 2020-01-13:
```
  this->_thingNameParameter = IotWebConfParameter("Thing name", "iwcThingName", this->_thingName, IOTWEBCONF_WORD_LEN);
  this->_apPasswordParameter = IotWebConfParameter("AP password", "iwcApPassword", this->_apPassword, IOTWEBCONF_WORD_LEN, "password");
  this->_wifiSsidParameter = IotWebConfParameter("WiFi SSID", "iwcWifiSsid", this->_wifiSsid, IOTWEBCONF_WORD_LEN);
  this->_wifiPasswordParameter = IotWebConfParameter("WiFi password", "iwcWifiPassword", this->_wifiPassword, IOTWEBCONF_WORD_LEN, "password");
```

Problematic (because hardcoded) and having confusing strings:
- "Thing name" (we want to use "Geiger accesspoint SSID")
- "AP password" (we want to use "Geiger accesspoint password")
- "WiFi SSID" (we want to use "WiFi client SSID")
- "WiFi password" (we want to use "WiFi client password")

Reasons:
- we do not want to confuse the users of our ESP32-based Geiger Counter, so we need to be clear about what needs to be entered in that form (**without needing addtl. docs** [who reads docs...])
- "Thing name" is just confusing if what one really wants to know is the ESP32 accesspoint SSID.
- We preferred "Geiger" over "ESP32" because our users know they have a geiger counter, but maybe not that it is based on ESP32 (we have non-technical users).
- We have 2 WiFis here, so we need to be clear when the ESP32 builtin **AP** is meant and when we configure the WiFi **client** that runs on the ESP32 (and needs the SSID/password to connect)

So, can we have these strings configurable, please?

ThomasWaldmann, 2020-01-13:
For a quick hack (just changing the hardcoded strings), this would be already an improvement (clarity):

- "Thing name" --> "WiFi accesspoint SSID"
- "AP password" --> "WiFi accesspoint password"
- "WiFi SSID" --> "WiFi client SSID"
- "WiFi password" --> "WiFi client password"

prampec, 2020-01-13:
You can access internal parameters directly by e.g. getThingNameParameter() where you can overwrite the 'label' with you designated string.

closed 2020-01-13 by prampec

ThomasWaldmann, 2020-01-13:
Good hint, I'll try.

But still you could make the defaults more clear, see above.

prampec, 2020-01-13:
Thing name is also used e.g. in host name. I use the word "WiFi" for station mode only, to keep things clear. I do not believe, that word "client" would help users understanding the situation.

ThomasWaldmann, 2020-01-13:
There are wifi accesspoints, there are wifi clients, so your usage is a bit specially and non-standard maybe?

And accesspoint vs. client helps at least more than what you have.

But anyway, do what you want and thanks for your hint, it worked. \o/

```
  iotWebConf.getThingNameParameter()->label = "WiFi accesspoint SSID";
  iotWebConf.getApPasswordParameter()->label = "WiFi accesspoint password";
  iotWebConf.getWifiSsidParameter()->label = "WiFi client SSID";
  iotWebConf.getWifiPasswordParameter()->label = "WiFi client password";
```