prampec/IotWebConf #99: Detecting wifi loss during Loop...

richiejarvis, 2020-01-14:
Hi - Thanks for this great library.  I've just started playing, and would like to force the ESP32S to reboot automatically if it is not connected to wifi.  I looked at the src and saw that in line 971 you are returning a boolean, not a bool.  See https://github.com/arduino/Arduino/issues/4673

I'll gen a Pull Request once I've sussed it out.  I was trying to do this in my code:

```
  if (iotWebConf.checkWifiConnection) {
    // do something
  }
```

Is there a better/easier way that you are aware of?

Cheers,

Richie

Line: https://github.com/prampec/IotWebConf/blob/7745fcb8e7e46c2be1e90012592d36092e90aa15/src/IotWebConf.cpp#L971

prampec, 2020-01-14:
Thank you for your comment. I always have used boolean as I always thought, that both bool and boolean are just aliases for uint8.

See method setWifiConnectionHandler() for detecting connection failures. Example: IotWebConf09CustomConnection

richiejarvis, 2020-01-14:
Thank you for the quick reply.  Yeah, it threw me out till I found that post!  I also discovered my error as well - I had not included the all important `()` after the function call.  It worked then.

Feel free to ditch the PR - it isn't needed now I've fixed my code.  It was an interesting tangent for me personally though.  

Soz to bother...

Cheers,

Richie

closed 2020-01-24 by richiejarvis