prampec/IotWebConf #103: Firmware update

pfeno, 2020-01-23:
Hi, 
I update the library to latest version and when i want to update the firmware I have the message below:
> File Not Found
> URI: /firmware
> Method: GET
> Arguments: 0

and on debug serial:
> Requested non-existing page '/firmware' arguments(GET):0

On my code, juste remove __skipApStartup()_ fix the issue but when I try to do the same on your sample (**IotWebConf04UpdateServer**, adding or not the method), its work !

Can I have some help!

prampec, 2020-01-23:
Yes! This is a known bug, and was already fixed in the main git branch. Please confirm that by switching to git version while testing?!

pfeno, 2020-01-24:
ok for me!
thanks.
I close the issue

closed 2020-01-24 by pfeno

prampec, 2020-01-24:
Great! So at least someone has tested the git version. Now I'm brave enough to publish a new release with the pending modifications.