prampec/IotWebConf #105: Feature request : OTA  auto update using Github bin file

happytm, 2020-01-24:
I think it will be useful addition to this library to have OTA auto update using .bin file on specific github repository. Iam using this feature here: 

https://github.com/happytm/BatteryNode/blob/master/ProbeSender.ino

````#include <ESP8266WiFi.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define binFile "https://raw.githubusercontent.com/happytm/BatteryNode/master/sender.bin"
#define BOOT_AFTER_UPDATE    false
HTTPClient http;

void otaControl() 
{

  rtcData.data[8] = pinNumber;      // Save device mode (0 for regular, 1 for autupdate and 2 for AutoConnect).
  
  Serial.print("Device Mode set to: ");
  Serial.println(rtcData.data[8]); 
  if (rtcData.data[8] == 1) 
  {  
  Serial.println("Start WiFi Connection......");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.println(".");
  }                                   
  Serial.println("Connected to WiFi");
  pinMode(LED_BUILTIN, OUTPUT);
  WiFiClient client;
  ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);

  ESPhttpUpdate.rebootOnUpdate(BOOT_AFTER_UPDATE);
  Serial.println("New firmware will be loaded now..............");
  t_httpUpdate_return ret = ESPhttpUpdate.update(binFile,"","CC AA 48 48 66 46 0E 91 53 2C 9C 7C 23 2A B1 74 4D 29 9D 33");
  rtcData.data[8] = 0;     // Set device mode (0 for regular, 1 for autupdate and 2 for AutoConnect).

  Serial.print("Device Mode set to: ");
  Serial.println(rtcData.data[8]); 
  
  http.end();

    switch(ret) {
         case HTTP_UPDATE_FAILED:
            Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
           // rtcData.data[8] = 2;            // Save device mode (0 for regular, 1 for autupdate and 2 for AutoConnect).
            rtcData.data[8] = 2;   // Set device mode (0 for regular, 1 for autoupdate and 2 for AutoConnect).
            Serial.print("Device Mode set to: ");
            Serial.println(rtcData.data[8]);  
            break;

         case HTTP_UPDATE_NO_UPDATES:
            Serial.println("HTTP_UPDATE_NO_UPDATES");
            rtcData.data[8] = 2;     // Set device mode (0 for regular, 1 for autupdate and 2 for AutoConnect).
            Serial.print("Device Mode set to: ");
            Serial.println(rtcData.data[8]);  
            break;

         case HTTP_UPDATE_OK:
            Serial.println("HTTP_UPDATE_OK");
            delay(5000);                    // wait for few seconds issue command with payload 36/09/00/.
            Serial.print("Device Mode set to: ");
            Serial.println(rtcData.data[8]); 
            rtcData.data[8] = 0;
            ESP.restart();
            break;

         default:
            Serial.printf("Undefined HTTP_UPDATE function: ");Serial.println(ret);
              }

    } else if (rtcData.data[8] == 2)
    
    { 
            
      
    } else {
     
      
     }
  }

Thanks.

prampec, 2020-01-28:
OTA is a great thing, but does not strictly connect to IotWebConf. So I don't plan to integrate it with IotWebConf.
But you are right, that an example usage with IotWebConf of it could be very useful.

RaviHotkar, 2021-06-12:
HTTP_UPDATE_FAILD Error (-1): HTTP error: connection failed
We are getting a Error of above line. How to solve it?

RaviHotkar, 2021-06-12:
@happytm : We are following exactly the procedue like the code you showed and also uploading the code from Github . Any special care to be taken in Github ?

happytm, 2021-06-12:
I  wrote this so long ago and since then I changed the name of file to "sender.ino.d1_mini.bin" instead of "sender.bin" to make it easy to upload compiled binary to github without changing name.

Hopefully that will solve the problem.