prampec/IotWebConf #111: Add method to get the last EEPROM index

jlgoasis, 2020-02-04:
In my things applications I need to save(write) and load(read) some parameters from Schedule/Alarms to the internal ESP EEPROM. Therefore I need to know the latest EEPROM index used in your library.

Add a method to obtain it will be welcome

jsolla, 2020-05-24:
Instead of getting the last EEPROM address you could just provide an offset for the first parameter saved by IOTWEBCONF
Open IotWebConf.h and look for this:

```
#ifndef IOTWEBCONF_CONFIG_START
# define IOTWEBCONF_CONFIG_START 0
#endif

```
The problem is how to properly define this value from your .ino file. :/

kscholty, 2021-04-30:
> Instead of getting the last EEPROM address you could just provide an offset for the first parameter saved by IOTWEBCONF
> Open IotWebConf.h and look for this:
>
> ```
> #ifndef IOTWEBCONF_CONFIG_START
> # [...]

This can easily be achieved at least under PlatformIO by adding

build_flags =
  -DIOTWEBCONF_CONFIG_START="YOUROFFSET"

to platformio.init.
See the HackingGuide.md in the doc subdirectory