prampec/IotWebConf #113: RTOS SUPPORT

jhortwo, 2020-02-26:
Hi, I want to say than you have done a grate work with this library. The one thing that I can't doit with the library is make it work with RTOS. When I create a Task, just reboot the ESP32 constanly. I need this feature to read onewire temperature sensors that are time sensitive. If you know other way to doit please let me know. Thanks.

richiejarvis, 2020-03-05:
I have also had the same issue, so will be investigating further on this also...

Currently trying to hunt a memory leak atm - hope it is in my code!

richiejarvis, 2020-03-05:
(the leak was in my code :) - easier to debug!)

prampec, 2021-04-14:
There was some stability fixes in v3.0.0. Please verify any unexpected problems against this version!