prampec/IotWebConf #116: web server port change

SanjayChakrapani, 2020-03-30:
First of all thank you for this library, this library is much more reliable, easy to understand and mainly NON-BLOCKING than any other WiFi configuration library.

I could not change the web server port from 80 to any other port, i even tried changing  dnsServer port and inside the library also, but nothing is working.
If I change the port number, the server is redirecting to the default port(80) but noting is showing(404)  

i tried to change here:
WebServer server(80);

and also here:
line 154  in IotWebConf.cpp file
MDNS.addService("http", "tcp", 80);

server.begin(8080) will also not work

societyofrobots, 2021-03-18:
I'm having this same exact problem.

Did you ever solve it?

If I use any port other than 80:
`WebServer server1(81);`

And then go to 192.168.10.117:81, I get redirected to 192.168.10.117 with a "192.168.10.117 refused to connect" error message.

I wonder if [WebServer.h](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/WebServer.h) is relevant, specifically these lines:
```
  WebServer(IPAddress addr, int port = 80);
  WebServer(int port = 80);
```

prampec, 2021-04-16:
Fixed. Will be available from v3.0.2

closed 2021-04-16 by prampec