prampec/IotWebConf #117: Wifi reconnection issue

SanjayChakrapani, 2020-03-30:
I'm using this library for so many days in my WiFi devices. The problem is when ever there is no WiFi available for too long time(5hrs) even after the connection is available ESP wont reconnect. only if the ESP is reset or esp put into forceApMode() and back to normal mode it will Reconnect 

I have also increased iotWebConf.setWifiConnectionTimeoutMs(ONE_MIN);
and reduced iotWebConf.setApTimeoutMs(3000);

an other of question 
Is there any option to skip AP mode completely when there is no wifi connected, only way should be forceApMode(true).
I have tried increasing WifiConnectionTimeoutMs but, as i told problem is it wont connect back aster long waiting time

iotWebConf.skipApStartup();  this only works in startup

Thank you :)

giovanni-foorban, 2020-06-16:
any updates on this issue? @prampec

prampec, 2021-04-14:
There are some stability problems fixed with v3.0.0, this might also fix strange/unexpected behaviors, it is worth to try. Please provide some feedback on this.
If the problem still exist, please try the same with plain ESP WiFi commands not using IotWebConf at all. This very much can be a bug in the ESP framework.