prampec/IotWebConf #120: Password not hidden on serial after config update

Miau09, 2020-05-16:
I have started testing this library in my code, and when configuring a WiFi connection on the config page, it will print the password for the WiFi SSID and the AP in plain text, despite the #define for visibility (I have removed my PW below).

14:27:53.189 -> Updating configuration
14:27:53.189 -> Value of arg 'iwcThingName' is:testAP
14:27:53.222 -> iwcThingName='testAP'
14:27:53.222 -> Value of arg 'iwcApPassword' is:REMOVED!!!
14:27:53.222 -> iwcApPassword was set
14:27:53.222 -> Value of arg 'iwcWifiSsid' is:ASUS
14:27:53.222 -> iwcWifiSsid='ASUS'
14:27:53.222 -> Value of arg 'iwcWifiPassword' is:REMOVED!!!
14:27:53.222 -> iwcWifiPassword was set
14:27:53.222 -> Saving config 'iwcThingName'= 'testAP'
14:27:53.222 -> Saving config 'iwcApPassword'= <hidden>
14:27:53.222 -> Saving config 'iwcWifiSsid'= 'ASUS'
14:27:53.222 -> Saving config 'iwcWifiPassword'= <hidden>
14:27:53.256 -> Saving config 'iwcApTimeout'= '30'

Now I am not sure how to fix this by just using some # ifdef IOTWEBCONF_DEBUG_PWD_TO_SERIAL
I think you need to check inside the IotWebConf::readParamValue() if we read a password, and if yes hide it ?
I hope this helps somehow.

prampec, 2021-04-14:
With v3.0.0 a complete rewrite fixed all similar problems. Please also check the Hacking guide (https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#compile-time-configuration) for compile time configuration.

closed 2021-04-14 by prampec

puyiot, 2021-05-21:
IotWebConfSettings.h
//Hidden password
// -- Logs passwords to Serial if enabled.
//#define IOTWEBCONF_DEBUG_PWD_TO_SERIAL

//Show password
// -- Logs passwords to Serial if enabled.
#define IOTWEBCONF_DEBUG_PWD_TO_SERIAL