prampec/IotWebConf #121: EEPROM kept open all times until config saved

Miau09, 2020-05-19:
Hey,

I have a startup procedure, where I write and read stuff from the EEPROM (e.G. if there is no wifi).

The problem now is, that if there is no valid configuration found by your libraray, the EEPROM will stay open forever as far as I understand it. It is only closed (commit), when the config is saved.

I do not understand why you don't close the eeprom when you have failed to find a valid config, and then simply open it again when you save the config. Or generaly speaking, do not keep the EEPROM open when you have done reading from it.

I am unable to use the EEPROM, because if I do so (read and write some offline parameters), I also close it when I am done.
Once it is closed, your config save function cannot save it again, because it not open anymore.

prampec, 2020-05-22:
In my understanding the the EEPROM.begin was needed to initialize the EEPROM usage and must be called at the beginning of every program uses EEPROM. I did not know about freeing up the EEPROM when it is in use.
Thank you, to pointing that out! I will definitely check this.

Miau09, 2020-05-26:
This whole eeprom thing is not so easy to understand, as there is no good documentation imho.
I still don't understand 100% how it is working, but I think you do not need to "free" stuff.

Main problem for me was, that I call eeprom.begin (100) in my code, when there is no WiFi.
I have an offset of 150 for your IoT, so if I call my begin function I overwrite the buffer, and your values are not existing anymore (since you never call begin again).

I have "fixed" this by simply calling begin(512) in my function, this includes your IoT length, and if your lib calls eeprom.commit(), everything should be fine.

I think it would be best that you do not rely on the once opened eeprom, but instead call eeprom.begin every time you want to read stuff from it?

prampec, 2020-12-20:
The EEPROM handling is rewritten in latest commits. Changes will be available with version v3.0.0

closed 2020-12-20 by prampec