prampec/IotWebConf #123: Execute normal code after configuration.

SensaIoTech, 2020-05-21:
Hello,

I'm new in use this amazing tool, and I'm doing a simple esp-32 code, where it read some ports status and send it over the web or send when the port status has changed.

The problem is, after the wi-fi setup I cannot execute the code, I've inserted it on the loop section, but it doesn't work.

Anyone can share an example of how to use a code after all the setup?

Thanks!

sgilissen, 2020-05-21:
Can you post your current code? 

I'm doing things in the loop section as well. 

Here's my loop section: 
`void loop() {
  iotWebConf.doLoop();

  // DHT sensor read loop
  unsigned long currentMillis = millis();
  if(currentMillis - previousMillis > DHT_INTERVAL) {
    previousMillis = currentMillis;
    dhtTmpC = dht.readTemperature();
    dhtTmpF = dht.readTemperature(true);
    dhtHumd = dht.readHumidity();
    
    
  } 
}`

SensaIoTech, 2020-05-21:
Hi,

This is basic code that I'm testing. Inside the loop.

`  iotWebConf.doLoop();  
    float humd = sht20.readHumidity();                  // Read Humidity
    float temp = sht20.readTemperature();               // Read Temperature
    Serial.print("Time:");
    Serial.print(millis());
    Serial.print(" Temperature:");
    Serial.print(temp, 1);
    Serial.print("C");
    Serial.print(" Humidity:");
    Serial.print(humd, 1);
    Serial.print("%");
    Serial.println();
    delay(1000);`

sgilissen, 2020-05-21:
I think I see the issue already. 
You're using a 1000ms delay. 
Delays are blocking. 
It's best to compare millis.

Try encapsulating your code in an if-statement the same way I did above. 
Don't forget to define the interval, and create the respective previousMillis = 0 outside your loop, above your setup() function.

So it would be something along the lines of:

```
// Define the interval to read the sensor
#define SHT20_INTERVAL 1000
int previousMillis = 0;

void setup() {

}

void loop() {
    iotWebConf.doLoop(); 
    
    // DHT sensor read loop
    unsigned long currentMillis = millis();
    if(currentMillis - previousMillis > DHT_INTERVAL) {
        previousMillis = currentMillis;
        float humd = sht20.readHumidity();                  // Read Humidity
        float temp = sht20.readTemperature();               // Read Temperature
        <print statements>
    }
    
}

```

Also, a small tip: I usually prefer to define a variable outside of any functions (eg. above the setup function), and just update these variables within the functions. 
That way they're global, and you can use them in pretty much any function.

SensaIoTech, 2020-05-21:
Thanks for the suggestion and tip, I will try this approach on the weekend!

prampec, 2020-05-22:
IotWebConf provides a delay() method, but I also recommend to compare millis instead: (current-start) > diff

SensaIoTech, 2020-05-26:
It's works! Nice and smoothly!

The only problem is when I set the difference to 10 minutes, it stops working, any suggestion what can be wrong?

prampec, 2020-05-26:
10 minutes is 10 * 60 * 1000 milliseconds, right?

SensaIoTech, 2020-05-26:
Yes, 10 min = 600000, but for some reason the Wifi changes the status and doesn't count.

lamello, 2021-02-04:
You could try to use typecasting, so instead of 600000 use 600000UL